ICube-Robotics/ethercat_driver_ros2 #9: EtherCAT upload can't upload nothing

enriLoniterp, 2022-07-26:
Hi @mcbed,
i am following the first part od the README before implementing ROS_control. After i started succesfully the master node i can't see my slaves with `ethercat slaves` as you mentioned whit this result:
`0 0:0 INIT+ 0x00000539:0x2200301
`
The problem is if i use:
`sudo ethercat upload -t uint16 0x1000 0x0`
it tells me:  `Failed to upload SDO: Input/output error`

Have you evere encountered this kind of problems?

thank you very much for your attention

mcbed, 2022-08-03:
Hi @enriLoniterp,
I never had this issue before, but as it is rather related to the [IgH EtherCAT](https://etherlab.org/en/ethercat/), maybe you could find some hints there.

enriLoniterp, 2022-08-08:
Hi @mcbed,
thanks in advice for helping me. 
I've written my user code starting from  [IgH EtherCAT](https://etherlab.org/en/ethercat/) /example/users and maybe i will solve this specific problem, anyway i decided to clone your code and directly try to build my first application and see if there are any problem.
Regarding your code i would like to ask you some questions. Should i clone it and add it to my ros_control code or directly develop from your github code?

Thank you, hope you have a good day
Enrico Andrini

mcbed, 2022-08-18:
Hi @enriLoniterp,
The driver is requires the main dependencies of ros2_conrol that need to be installed. Using it with the modules that are already implemented does not require any additional coding. If however you want to use another ethercat module, you need to create your own plugin. Feel free to create a PR to integrate it into our plugin collection. 
Once you have all plugins, you can assemble your applications by defining your plugins and interfaces in the ros2_conrol urdf file. Feel free to ask if you need additional information!

enriLoniterp, 2022-08-30:
Hi @mcbed,
My final task is to use my ros2 and moveit! project and combining it with this intergation. I would like to develop a simple pick and place application and part of my final target is to activate my modules with EtherCAT. 

At this moment i have a main project with Ros2 Humble and MoveIt! and building my application with my Comau Racer 5.080. I would like to combine my moveit application with this EtherCAT integration, this way i should have ROS2 application running with moveit! which give its final command to EtherCAT slaves. 

I've studied MoveIt and i've seen that it contains ROS2 Control, as mentioned in your README my EtherCAT Master is represented as Hardware Interface. i think that i should combine this two repository and try to translate MoveIt! moves in EtherCAT signals. But i DON'T understand  what i should need as ros2_controller to send a signal to EtherCAT slaves. 
Obviously this EtherCAT slaves should move a servo motor.

Enrico Andrini

mcbed, 2022-09-06:
Hi @enriLoniterp,
I think that what you want to do can be achieved using `chainable controllers` for the latest version of `ros2_control`. In this case you could use the classical `joint_trajectory_controller` to compute your targets and pass this to your custom controller that converts, let's say, positions into signal to send to your device.

enriLoniterp, 2022-09-06:
Hi @mcbed,
thank you very much!!

If i understood correctly, the solution might be:
1) use `joint_trajectory_controller` to receive data about trajectory 
2) use `chainable controllers` although `https://control.ros.org/master/doc/ros2_controllers/joint_trajectory_controller/doc/userdoc.html?highlight=chainable` (controller not implemented as chainable) but anyway i can take this ROS2 msg with relevant information
3) pass this informations to my controller that translate into signals to my robot **in the end** use your code of ethercat to send signals to my slaves

is that right?

Hope i'm not disturbing!
But thank you very much!
Enrico

mcbed, 2022-09-06:
That's the idea, yes :)

enriLoniterp, 2022-09-08:
Tank you very much @mcbed,
can i ask how to use your ros2_controller that i found in the forked repository?
I haven't really understood how to deal it properly.

Enrico

closed 2022-09-15 by mcbed