ICube-Robotics/ethercat_driver_ros2 #10: invalid tag ec_module

enriLoniterp, 2022-09-15:
hi @mcbed,
im trying your integration in my example code.
I've tried to follow your guide in READ.ME but i haven't the expected results.

1. I add `ethercat_driver_ros2` in my worskspace with my other packages to run my RViz simulation with my ethercat driver
2. Everything compiled, even if put another digital output, it is an `Beckhoff el2809` and i made its plugin following the others examples you previously add
3. i have installed master Igh and it is all ok with it (personally tested)
4. i had in my ros2_control file the the ethercat-driver as you explained
5. added EcModule in ros2_control description file
6. For interfacing controllers with EtherCAT Slave Modules i tried your example with an EtherCAT IO modules 
  <gpio bane="MYGPIO"> etc. etc

in 4. it gives me this error ros2_control
```
[ros2_control_node-5] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-5]   what():  invalid tag name ec_module
```
it seems it doesn't understand that it is Hardware resource

and when i don't delete 6. it literally don't start and stop with error on launch.

What is should modify?

enriLoniterp, 2022-09-15:
Thank you very much for your fast reply.

Consider i'm using MoveIt tutorials packages and i'm using a panda robot.
My final target is to turn on a  led on my BeckHoff EL2809 16-digital outputs through ROS 2.

To answer your question.
- Which branch are you using and which version of ROS2 ? main branch and i'm using ROS 2 Humble
- ```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:macro name="panda_ros2_control" params="name initial_positions_file">
        <xacro:property name="initial_positions" value="${load_yaml(initial_positions_file)['initial_positions']}"/>

        <ros2_control name="${name}" type="system">
            <hardware>
                <plugin>mock_components/GenericSystem</plugin>
            </hardware>

            <hardware>
                <plugin>ethercat_driver/Ethercatdriver</plugin>
                <param name="master_id">0</param>
                <param name="control_frequency">100</param>
            </hardware>

            <ec_module name="ECModule">
                <plugin>etherca_plugins/ECModule</plugin>
                <param name="alias">0</param>
                <param name="position">1</param>
           </ec_module>

            <gpio name="myGPIO">
            <command_interface name="dig_output.1"/></gpio>
            <ec_module name="EL2809">
                <plugin>etherca_plugins/Beckhoff_EL2809</plugin>
                <param name="alias">0</param>
                <param name="position">1</param>
                <param name="do.1">dig_output.2</param>
            </ec_module>
            </gpio> 

            <joint name="panda_joint1">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['panda_joint1']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
            <joint name="panda_joint2">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['panda_joint2']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
            <joint name="panda_joint3">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['panda_joint3']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
            <joint name="panda_joint4">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['panda_joint4']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
            <joint name="panda_joint5">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['panda_joint5']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
            <joint name="panda_joint6">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['panda_joint6']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
            <joint name="panda_joint7">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['panda_joint7']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
        </ros2_control>
    </xacro:macro>
</robot>
```

mcbed, 2022-09-15:
I don't know if this is the issue but 2 things:
1. I'm not sure this can work in ros2_control as you have here 2 plugins for the same system which I think will fail.
```xml
<hardware>
    <plugin>mock_components/GenericSystem</plugin>
</hardware>

<hardware>
    <plugin>ethercat_driver/Ethercatdriver</plugin>
    <param name="master_id">0</param>
    <param name="control_frequency">100</param>
</hardware>
```
2. Each ec_module is designed to be a child of a joint, gpio, or sensor
```xml
       <ec_module name="ECModule">
            <plugin>etherca_plugins/ECModule</plugin>
            <param name="alias">0</param>
            <param name="position">1</param>
       </ec_module>

```

mcbed, 2022-09-16:
Putting the 2 plugins inside a single ros2_control system should not work as it is against the logic of the framework. A ros2_control system is intended to represent a single communication channel. So here you have 2 of them: one for the panda and another for ethercat. You should then have 2 ros2_control systems in your description. 

In 2. you still have the closing `gpio` tag that exludes your ec_module from the component. 

For the last, in the `gpios` parameter you should specify the names of the gpios you want to control. In your case its `myGPIO`. 
Then, you specify the interface per listed gpio: 
```yaml
command_interfaces:
    myGPIO:
        - dig.output.1
```

enriLoniterp, 2022-09-16:
Hi @mcbed,
sorry for this question, i'm still learning `ros2_control`.

Thank you for the first suggestion, i should delete mock component and only test with `ethercat_drivers`, but i still don't know how to make a simple write with etherCAT master with your `ethercat_drivers`.

For example, i've created a simple application including `ros2_control` and your etherCAT drivers and i want to actually turn on a led on `Beckhoff EL2809`, in my experience it's just writing with **EC_WRITE8** from my `IgH master` or other simple macro or functions and the led will turn on. I want to do the same but included in a ROS2 application with your drivers. I've seen you published a simple `gpio controller` and i'm interested to make it part of my initial simple application. But what are the moves should i do? I think:

1. add my gpio 16 digital ouput beckhoff on plugin list? Done
2. add succesfully the `ethercat_drivers` to my project ros2? Done
3. write a node in which i instantiate your `gpiocontroller `and make a write to my `Beckhoff el2809`? from now i'm still a bit confused..
4. ..
5. ..

Hope you can help,
but thanks you a lot anyway!

Enrico

mcbed, 2022-09-16:
Did you write the plugin driver for your module inheriting from the `ethercat_interface::EcSlave` base class ?

In general you should:
1. write the plugin for your module if not already available in the [list](https://github.com/ICube-Robotics/ethercat_driver_ros2/blob/main/ethercat_plugins/available_plugins.md). Example for Beckhoff  20xx modules [here](https://github.com/ICube-Robotics/ethercat_driver_ros2/blob/main/ethercat_plugins/src/beckhoff_plugins/beckhoff_el20xx.cpp)
2. set up the description of your ec modules in your hardware components with `ec_module` tags and map the interfaces of ros2_control to the module channels. Also some modules reuire additional parameters that can be set in the description. This depends on how the plugin is written.
3. run the `gpio_command_controller` to interact with the interfaces that are mapped to your module channels. You done need a specific node, you can just run it from command prompt with `$ ros2 topic pub ... `

enriLoniterp, 2022-09-16:
1. i went on `ethercat_driver_ros2/ethercat_plugins/src/beckhoff_plugins` and i added` beckhoff_el2809.cpp` which is my 16 digital output gpio. (i also have on ek1100 you already prepared)
2.  i put in <ros2_control> my ec_module mapped correctly, not added any additional parameters for now, maybe i will. You can find the ros2_control sample if you did moveit2 tutorials on panda robot
3. i don't understand this point, i run gpio_command_controller using one `ros2 control `command or it is a node itself which i can run from bash?  by the way i'd rather prefer writing my node because this ethercat write should represent command sent to my beckhoff element which should move my robot theoretically!

mcbed, 2022-09-20:
Second, in your controller cofiguration you are missing the gpios see [doc](https://github.com/mcbed/ros2_controllers/blob/gpio_controllers/gpio_controllers/doc/userdoc.rst) : 
```yaml
gpio_command_controller:
  ros_parameters:
    gpios:
        - myGPIO
    command_interfaces:
        myGPIO:
            - dig_output.1
```

enriLoniterp, 2022-09-20:
Should work this  #way?
```
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz

    scara_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    scara_position_controller:
      type: position_controllers/JointGroupPositionController

    scara_joint_velocity_controller:
      type: scara_joint_velocity_controller/ScaraJointVelocityController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    gpio_command_controller:
      type: gpio_controllers/GpioCommandController
  

scara_trajectory_controller:
  ros__parameters:
    command_interfaces:
      - position
    state_interfaces:
      - position
    joints:
      - joint1
      - joint2
      - joint3

  scara_position_controller:
    ros__parameters:
      joints:
        - joint1
        - joint2
        - joint3

  scara_joint_velocity_controller:
    ros__parameters:
      joints:
        - joint1
        - joint2
        - joint3

  gpio_command_controller:
    ros__parameters:
        gpios:
          - myGPIO
        command_interfaces:
          myGPIO:
            - dig_output.1
        state_interfaces:
            myGPIO:
            - dig_output.1

```

> Second, in your controller cofiguration you are missing the gpios see [doc](https://github.com/mcbed/ros2_controllers/blob/gpio_controllers/gpio_controllers/doc/userdoc.rst) :
>
> ```yaml [...]

enriLoniterp, 2022-09-20:
i've cleaned my `ros2_control file` and divided in two pieces with <hardware> as you suggested above.
But with .yaml file in my previous comments i got
`
 [ros2_control_node-2] [ERROR] [1663669037.728825542] [controller_manager]: Can't activate controller 'gpio_command_controller': State interface with key 'myGPIO/dig_output.1' does not exist`

It seems control manager does not get what it expects. 

thank you again for your help,
Enrico

mcbed, 2022-09-20:
Looks good to me. Just the `state_interfaces` tag is useless as not used by the controller

enriLoniterp, 2022-09-21:
Hi @mcbed,

i agree but i remember it seems i had a problem making everything working without it, anyway i will fix it, for now i'm just happy is connecting!

In particular i would like to ask you some new information about ethercat drivers,
It seems to connect well, and i think i'm sending the right value to my `EL_2809 `usin `ros2 topic pub` but i'm not sure i understand every passage. three components are involved:

- `ethercat driver`: i think everything is OK, it finds mine one gpio module, i'm pretty sure it finds my slave also and finally `on_activate` method is succesfull
- `master ethercat` is working i assume
- `gpio controller`:  still don't fully understand, it exposes some topic used to write on the ethercat master, i've seen that value i passed through gpio controller is sent to processData method in BeckHoff plugin, then you use bit_masking on `dig_output` and i sent it on my domain, this method is the same of IgH master which i previously used. (in my app i wrote domain +1 and after my value).

Thank you very much for the great help! Your helping me a lot!

Enrico

enriLoniterp, 2022-09-26:
Hi @mcbed ,

Another thing i would ask you is how can i send repeated messagges to my BeckHoff2809 in order to maintain powered on the led. 
My solution was quite simple in `Igh EthercatMaster`:
```

void cyclic_task()
{

    ecrt_master_receive(master);
    ecrt_domain_process(domain1);

    // check process data state
    check_domain1_state();
    if (counter) {
        counter--;
    } else { // do this at 1 Hz
        counter = FREQUENCY;

        // calculate new process data
        blink = !blink;
        // check for master state (optional)
        check_master_state();
        // check for slave configuration state(s) (optional)
        // check_slave_config_states();
    }

#if 0
    // read process data
    printf("AnaIn: state %u value %u\n",
            EC_READ_U8(domain1_pd + off_ana_in_status),
            EC_READ_U16(domain1_pd + off_ana_in_value));
#endif

#if 1
    u_int8_t result;
    // write process data
    EC_WRITE_U8(domain1_pd + off_dig_out, blink ? 0x00 : 0x01);
    EC_WRITE_U8(domain1_pd + off_dig_out2, blink ? 0x00 : 0x08);
    result = EC_READ_U8(domain1_pd + off_dig_in);
    printf("input slave3 uint: %x\n", result);
    printf("input slave3 bit1: %c\n\n", result & 0x01);
#endif
    // send process data
    ecrt_domain_queue(domain1);
    ecrt_master_send(master);
}
```

As you can see i wrote periodically :
```
  EC_WRITE_U8(domain1_pd + off_dig_out, blink ? 0x00 : 0x01);
  EC_WRITE_U8(domain1_pd + off_dig_out2, blink ? 0x00 : 0x08);
```
the same i want to do with you plugin but it seems that ` proceed_data` in the plugin of my actuator gets error if include any cycles. Maybe i should put this cylce inside the `gpio_controller`?

mcbed, 2022-09-26:
Hi, 
It depends on how you did your plugin I suppose. In the case of our plugins, the value of the `state_interface` remains the same until a controller changes it so if you want your led to blink you need to publish 1/0/1/... in your topic.  You can do it either in a dedicated controller or using a ros2 node.

enriLoniterp, 2022-09-27:
Hi,
i'm using `ethercat_driver `plugin and the controller involved is `gpio_controller`, maybe i should change update method in the controller.

So, i don't change the controller, but i need a simple publisher like this:

```
 public:
    MinimalPublisher()
    : Node("minimal_publisher"), count_(0)
    {
      publisher_ = this->create_publisher<std_msgs::msg::String>("topic", 10);   
//using CmdType std_msgs::msg::Float64MultiArray;
      timer_ = this->create_wall_timer(                             
      500ms, std::bind(&MinimalPublisher::timer_callback, this));                           
    }
```

As you can see i stated using Float64MultiArray because it was included in your controller but actually i'm not interested, for my purpose it is only write a simple `int value`.
Should i change it and publish on simple int message? 

Thank you again,
Enrico

mcbed, 2022-09-27:
The message type for the topic is necessary a `Float64MultiArray`. If you want to publish a single `int` value just send an array of 1 `float` element and read it as an `int`.

mcbed, 2022-09-28:
Congrats ! I am happy you could make it work ! 
1. You can but this requires to develop your custom controller.
2. In my applications I write a message array of 16 floats and use the `gpio_controller` to pass them as `state_interfaces` to the ec module plugin that read them according to the mapping you pass as module parameters.

enriLoniterp, 2022-09-28:
Thank you!

I think i will built my custom controller, this code should work on a real robot one day! It should be something representing a real motor, i think maybe they should be more plugins which will receive one value!!

I don't understand something of you 2.. Why you said `state_interfaces`, you are refering to `command_interfaces`? I've always thought that state_interfaces is for reading from the `hardware_interfaces` while command to write.

mcbed, 2022-09-28:
> 
> I don't understand something of you 2.. Why you said `state_interfaces`, you are refering to `command_interfaces`? I've always thought that state_interfaces is for reading from the `hardware_interfaces` while command to write.

Sorry, too fast typing, you are right of course!

enriLoniterp, 2022-09-29:
hi @mcbed,
two questions:

1. My digital output has 2 blocks of 8bit output value:  };
```
...... // code
    ec_sync_info_t syncs_[3] = {
        {0, EC_DIR_OUTPUT, 8, pdos_ + 0, EC_WD_ENABLE},
        {1, EC_DIR_OUTPUT, 8, pdos_ + 8, EC_WD_ENABLE},
        {0xff}
    };
    DomainMap domains_ = {
        {0, {0} },     <-- first block
        {1, {8} }      <-- second block
    };
};

```
my addresses start at 0x7000 (first 8 bit) 0x7080 (second 8 bit)
i can't write properly on the second. How am i supposed to make 2 EC_WRITEU8 on this output?
consider i have assigned digital output in ros2 controller file!

2. I've noticed that if i sends every minutes and set `ros2_control` frequency at 2Hz , it still at 2Hz freq.. This mean ros2_control publish everytime and not just only one time. IIs it possible to write on gpio el2809 only one time?
3. And it is possible to have a feedback mechanism from the Bechoff gpio?

Thank you again @mcbed

mcbed, 2022-09-29:
hi,
> 1. My digital output has 2 blocks of 8bit output value:  };
>
> ```
> ...... // code
>     ec_sync_info_t syncs_[3] = {
>         {0, EC_DIR_OUTPUT, 8, pdos_ + 0, EC_WD_ENABLE},
>         {1, EC_DIR_OUTPUT, 8, [...]
what are your  `channels` and `pdos` ? I think the problem comes from the way you configure the `domains`.
>     2. I've noticed that if i sends every minutes and set `ros2_control` frequency at 2Hz , it still at 2Hz freq.. This mean ros2_control publish everytime and not just only one time. IIs it possible to write on gpio el2809 only one time?
I do not understand the problem here, sorry. Isn't the freq what you set ? 
>     3. And it is possible to have a feedback mechanism from the Bechoff gpio?
This depends on the module I suppose. You would need to check if the register value fits what you want. Alternatively, in the plugin you can define a `state_interface` that takes the value of the `command_interface` just after writing to the module. This is something we do [here](https://github.com/ICube-Robotics/ethercat_driver_ros2/blob/77232946925e4e8da9d61349e2adb94c625ab6e4/ethercat_plugins/src/beckhoff_plugins/beckhoff_el21xx.cpp#L35)

enriLoniterp, 2022-09-29:
1. first problem solved!, just written 16 and set one domain, now it's working!

2. My module is able to send back a "status_word" (something similar to an _ACK_). I will study that plugin.

As always, thank you very much!

closed 2022-10-10 by mcbed