ICube-Robotics/ethercat_driver_ros2 #11: /dev/EtherCAT0 can't start without permission

enriLoniterp, 2022-09-20:
Hi @mcbed,

When you try to activate your master you may encounter few problems:

1. You haven't started your master and ros2-control might say `No such file or directory ``/dev/EtherCAT0`
2. If you started with `/etc/init.d/ethercat start` it might say `Permision denied`

If you use `/etc/init.d/ethercat start` and then `sudo chmod /dev/EtherCAT0 ` (don't know exactly minimum permission required)
This way should work.

Enrico

mcbed, 2022-09-20:
Normally you don't need to do it. Is your `$USER` member of `ecusers` ? You can run  `$ groups` in a shell to check it.

enriLoniterp, 2022-09-20:
yes, it is but it doesn't allow me to connect!

My current user is in the ecusers group

closed 2022-09-27 by mcbed

ros2torial, 2022-10-17:
you can also use **sudo -s** or **sudo su** before running the commands to give all the permissions and hardware access.