ICube-Robotics/ethercat_driver_ros2 #25: Qeustion: Velocity controller for maxon_epos3

omarhogni, 2023-02-01:
What controller do you use for velocity mode on maxon_epos3?

I am trying to implement a cia 402 compatible VFD but am having issues interfaceing with the exposed
hardware interface.

Finding valid controller configurations and examples is not trivial it seems.

mcbed, 2023-02-09:
hi @omarhogni, 
There is currently a non intuitive remapping system for interfaces that will be removed in PR #28 . With these minor changes, you should be able to interface your drive using the `velocity_controllers/JointGroupVelocityController` and use as `ros2_control` description:
```xml
<ros2_control name="ec_single_axis" type="system">
  <hardware>
    <plugin>ethercat_driver/EthercatDriver</plugin>
    <param name="master_id">0</param>
    <param name="control_frequency">100</param>
  </hardware>

  <joint name="joint_1">
    <state_interface name="position"/>
    <state_interface name="velocity"/>
    <state_interface name="effort"/>
    <command_interface name="position"/>
    <command_interface name="velocity"/>
    <command_interface name="effort"/>
    <ec_module name="Maxon_EPOS3">
      <plugin>ethercat_plugins/Maxon_EPOS3</plugin>
      <param name="alias">0</param>
      <param name="position">0</param>
      <param name="mode_of_operation">8</param>
    </ec_module>
  </joint>

</ros2_control>

```

omarhogni, 2023-02-09:
Thank you. I got i running.

closed 2023-02-09 by omarhogni