ICube-Robotics/ethercat_driver_ros2 #34: Ethercat faults and suggestions

GiladRobee, 2023-02-19:
system: Ubuntu 22.04
drivers: Thecnosoft IPOS 4808
1. there is a visible jitter in the motion of the motor that Phillipe thought was from the synchronization of the EtherCAT master with the drivers. </br>
solution:
we have dedicated a specific CPU core for the EtherCAT master and the motion runs much smoother
2. permission errors each time the machine is booted/EtherCAT master was restarted </br>
no solution was found at the moment besides ```$sudo chmod 664 /dev/EtherCAT0``` each time
3. there is a need for homing operation process to interact with the driver homing or with external limit switches
4. there is a need to pass the product_id to the pluggin as a param since different Thecnosoft drivers have different product id</br>
solution: we can duplicate the pluggin for each product id but that does not seems to be the optimal solution

mcbed, 2023-02-20:
Hi @GiladRobee,
point 4 was just fixed in PR #33.

GiladRobee, 2023-02-21:
when running on ARM64 ./bootstrap fails because of autoconf in order to fix this automake needs to be installed insead.
after running the download we have encountered error 2 (the permission error) again

mcbed, 2023-02-21:
@GiladRobee, you can try giving ethercat access to all users not only `ecusers` by running this line : 
```shell
$ sudo bash -c "echo KERNEL==\"EtherCAT[0-9]*\", MODE=\"0664\" > /etc/udev/rules.d/99-EtherCAT.rules"
$ sudo udevadm control --reload-rules
$ sudo udevadm trigger
```

Does it change something ?

GiladRobee, 2023-02-22:
Unfortunately no,
the same error still occurs.

Brethiltaras, 2023-02-23:
Hi, about issue 2, I found in IGH EtherCAT master 1.52 documentation, section 9.5
 
"
The device nodes will be created with mode 0660 and group root by default. If “normal” users shall have reading access, a udev rule file (for example /etc/udev/rules.d/99-
EtherCAT.rules) has to be created with the following contents:
KERNEL ==" EtherCAT [0 -9]*" , MODE ="0664"
"

Brethiltaras, 2023-02-23:
Hi, about point 2, I found in the IGH document 9.5:

The device nodes will be created with mode 0660 and group root by default. If “normal” users shall have reading access, a udev rule file (for example /etc/udev/rules.d/99-EtherCAT.rules) has to be created with the following contents:
KERNEL ==" EtherCAT [0 -9]*" , MODE ="0664"

It seems that the double quotes are missed in the default-generated one.

GiladRobee, 2023-02-23:
Thank you for the input, I have changed the rules and reloaded them but the
same issue still exists.

Brethiltaras, 2023-02-24:
> Thank you for the input, I have changed the rules and reloaded them but the same issue still exists.
> […](#)

What about 
KERNEL=="EtherCAT[0-9]*", MODE="0664", GROUP="ecusers"

mcbed, 2023-02-27:
finally it should be `KERNEL ==" EtherCAT [0 -9]*" , MODE ="0666"` in this case

closed 2023-02-28 by GiladRobee