ICube-Robotics/ethercat_driver_ros2 #54: Example of ethercat_generic_plugins/EcCiA402Drive

jaronski, 2023-04-13:
Hello,
would it be possible to show an example of how to integrate the EcCiA402Drive into a xacro and the corresponding config.yaml?
i have a CiA402 Drive that i would like to use with the ethercat_driver_ros2 but i am having trouble with the configuration. I adjusted the Configuration File as shown in the MaxonEpos3 example to the appropriate values to match my drive (rpdo, tpdo). I created a  urdf.xacro with the ros2_control tag where the ethercat_driver is started and configured 1 joint with command interface position and velocity as ec_module i choose the plugin ethercat_generic_plugins/EcCiA402Drive put the alias and position on the ethercat bus mode_of_operation to 8 or 9 depending on what i want to test out and the param slave_config i put the config.yaml i adjusted. 

how can i claim the   <command_interface name="reset_fault"/> and to what controller? Can you show this for the EPOS3?
i am able to start the drive and move it to a position sometimes and then i restart and when i launch everything again sometimes the drive doesn't get into Operation Enabled and i don't understand why. 

Any help or pointers in the right direction would be much appreciated
best regards
jaronski

jaronski, 2023-04-14:
Hi @mcbed 
thanks for your response. I will check that out first thing on Monday 😉

best regards and a great weekend

AlessioMosca, 2023-04-21:
> > 
Hi @mcbed,
may i ask why he needs to add <state_interface name="reset_fault"/>  instead of <command_interface name="reset_fault"/> 
?

Best Regards
Alessio

mcbed, 2023-04-21:
Hi @AlessioMosca,
You need both for the controller to configure. It is a incomplete feature but I did not have the time to fix it yet.

qleonardolp, 2025-04-30:
Dear, @mcbed 
About the fault reset feature, would it be a problem to use the `control_word` command interface directly to perform the fault reset, writting `0x80`, or 128 on it?

I'm writting a custom controller, based on the joint trajectory controller, and I'm interested on have a controller able to do fault handling.