ICube-Robotics/ethercat_driver_ros2 #57: colcon build std:error

AlessioMosca, 2023-04-21:
Hi there,

I am following the instruction on how to compile the pkg by colcon build:
`colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --symlink-install`

and  get the following output:
`5 packages had stderr output: ethercat_driver ethercat_generic_cia402_drive ethercat_generic_slave ethercat_interface ethercat_plugins`

However, after I get the error, if I reuse the command previous command I am able to compile without errors.

Any suggestion ?
Can be a dependency problem ?

Bets regards 
Alessio

datamomo, 2023-04-21:
Hi Alessio,

even though I'm not an expert I use colcon sometimes to build packages. I experience that kind of output very often. It does not mean that your build process failed. If that would be the case the output would be something like: 
```
Summary: x package(s) finished [time]
  x package(s) failed: package_xy, ...
  ...
```
For me personal the output you got never caused any problems with using the built packages. I think the output can also be related to warnings and information the packages provide via command line during build. Anyways if you want to see more clearly what happens I think you can try something like:
```
colcon build --event-handlers console_direct+
```
This should monitor all your build output in realtime.

Best regards

AlessioMosca, 2023-04-21:
Hi @datamomo,

Many thanks

Best Regards
Alessio

closed 2023-04-21 by mcbed