ICube-Robotics/ethercat_driver_ros2 #64: Can't get the rosdep to install

practical-mechanics, 2023-05-03:
Hi!
I'm trying to install the driver, but I got stuck at this command:
`rosdep install --ignore-src --from-paths . -y -r`
Getting this error:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ethercat_driver_ros2: Cannot locate rosdep definition for [ethercat_plugins]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
``` 
Did I do something wrong in the process?
Thanks!

datamomo, 2023-05-04:
Hi,

I think this isn't your fault. As you can see [here ](https://github.com/ICube-Robotics/ethercat_driver_ros2/commit/0fb9fc4a7c92331dc2d15b9beac3c45098d01870) previously there was one colcon package for all plugins called "ethercat_plugins". This package does not exist anymore since the plugins were split into vendor specific packages.
This error occurs because the "ethercat_plugins" package is still referenced in the [package.xml](https://github.com/ICube-Robotics/ethercat_driver_ros2/blob/main/ethercat_driver_ros2/package.xml). So since this package does not exist anymore you can ignore that error. For me it did not cause any problem. But if you want to remove the error completely I think you just can remove the dependency from the previously mentioned package.xml.

Best regards

practical-mechanics, 2023-05-04:
All right, thank you!

closed 2023-06-06 by mcbed