ICube-Robotics/ethercat_driver_ros2 #69: yaml-cpp linking error on build

mauricesvp, 2023-06-25:
Following the install instructions, I get the following error after `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --symlink-install`:

```bash
Starting >>> ethercat_interface
Starting >>> ethercat_msgs
Finished <<< ethercat_interface [0.29s]                                                                
Starting >>> ethercat_driver
Starting >>> ethercat_generic_slave
Starting >>> ethercat_advantech_modules                                                                                      
Finished <<< ethercat_driver [0.32s]                                                                                                                                                          
Starting >>> ethercat_ati_modules
Finished <<< ethercat_advantech_modules [0.35s]
Starting >>> ethercat_beckhoff_modules
--- stderr: ethercat_generic_slave                                                                                                                                                                
/usr/bin/ld: /usr/local/lib/libyaml-cpp.a(scanner.cpp.o): warning: relocation against `_ZNSt16allocator_traitsISaIN4YAML7Scanner11FLOW_MARKEREEE8allocateERS3_m' in read-only section `.text._ZNSt11_Deque_baseIN4YAML7Scanner11FLOW_MARKERESaIS2_EE16_M_allocate_nodeEv[_ZNSt11_Deque_baseIN4YAML7Scanner11FLOW_MARKERESaIS2_EE16_M_allocate_nodeEv]'
/usr/bin/ld: /usr/local/lib/libyaml-cpp.a(exceptions.cpp.o): relocation R_X86_64_PC32 against symbol `_ZTVN4YAML9ExceptionE' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/ethercat_generic_slave.dir/build.make:199: libethercat_generic_slave.so] Fehler 1
gmake[1]: *** [CMakeFiles/Makefile2:172: CMakeFiles/ethercat_generic_slave.dir/all] Fehler 2
gmake: *** [Makefile:160: all] Fehler 2
---
Failed   <<< ethercat_generic_slave [0.56s, exited with code 2]
Aborted  <<< ethercat_ati_modules [0.24s]
Aborted  <<< ethercat_beckhoff_modules [0.20s]
Aborted  <<< ethercat_msgs [0.90s]                              

Summary: 3 packages finished [1.20s]
  1 package failed: ethercat_generic_slave
  3 packages aborted: ethercat_ati_modules ethercat_beckhoff_modules ethercat_msgs
  1 package had stderr output: ethercat_generic_slave
  6 packages not processed
```

So there seems to be some problem with yaml-cpp/yaml_cpp_vendor. Has anyone encountered this before? Any guesses on how to fix this?

Thanks in advance.

mauricesvp, 2023-06-26:
Build yaml-cpp from [source](https://github.com/jbeder/yaml-cpp), I don't get the error. Maybe the issue is a wrong/old version in the ros2 [yaml-cpp vendor package](https://github.com/ros2/yaml_cpp_vendor/).

mcbed, 2023-06-28:
I think it can be an issue related with your setup (maybe outdated libs ?) as CI is running just fine.

mauricesvp, 2023-06-28:
> I think it can be an issue related with your setup (maybe outdated libs ?) as CI is running just fine.

Most likely this is the case, closing this issue.

closed 2023-06-28 by mauricesvp