ICube-Robotics/ethercat_driver_ros2 #84: The motor won't run in "Profiled_Position" mode when using "generic_ec_cia402_drive"

ZorroStardust, 2023-10-12:
I have already successfully made the motor run in "Cyclic_sync_position" mode, which is the defaut mode. Now I'm trying to run in "Profiled_position" mode, so I wrote "<param name="mode_of_operation">1</param>" in the xacro file, and the motor switched on in "Profiled_position" mode indeed. But when I run the command_publisher which send the target_position to the controller, the motor didn't move, I checked the target position of the motor and found that the value didn't change into the value I sent. This is the first question.

And the second question is that, we know that in "Profiled_position" mode, we need to write the control_word"0x1f" after writing a new target position to the motor to enable it, but I didn't find this process in the "generic_ec_cia402_drive", does it mean I need to modify the code ? If so, how am I suppose to do it?

CaPlasberg, 2023-10-30:
Had the same issue. For now, I'm not happy with this proposed solution. Changing line 63/64 in generic_ec_cia402_drive.cpp helped:
`pdo_channels_info_[index].override_command = false`
This basically always writes the value of the position interface to the corresponding register, regardless of which mode you are using. Probably, adding the PROFILED_POSITION Mode to the exceptions to not overwrite the position value would also solve your problem.

mcbed, 2023-10-31:
Hi, the driver was in fact intended to work in cyclic sync mode, as it is the best suited mode for ros2_control. Any contribution to make it more generic is however welcome !

ZorroStardust, 2023-11-07:
I have already solved my problem, I was too naive that I thought the "generic_ec_cia402_drive" includes all mode of operation.
I later modified the code so that it could run in profiled_position mode. I will post my code here later. Thanks!

> Hi, the driver was in fact intended to work in cyclic sync mode, as it is the best suited mode for ros2_control. Any contribution to make it more generic is however welcome !

JensVanhooydonck, 2023-12-08:
> I have already solved my problem, I was too naive that I thought the "generic_ec_cia402_drive" includes all mode of operation. I later modified the code so that it could run in profiled_position [...]

Hi, can you post your solution?

JensVanhooydonck, 2023-12-12:
I've added a PR which adds a basic form of PP-mode. https://github.com/ICube-Robotics/ethercat_driver_ros2/pull/95 

This sets the New Target Position bit when receiving a new value in PP-mode. And also reset's this afterwards.