ICube-Robotics/ethercat_driver_ros2 #100: control motors to move

wjx1991, 2023-12-23:
Hi,
I have six slaves with each driving a motor.  The igh master is created and the motors are enabled after running the code. Next, I want to make the motor rotate. Could you please provide an example or some instructions on how to do that?

jaronski, 2024-01-12:
Hi wjx1991,
I guess you ran into the same problem i had with the ethercat_driver_ros2. Motor is slowly moving out of its default position? As i investigated while debugging the ethercat_driver i noticed that when starting the system inside of the code (// setup current position as default position) was getting called over and over again updating the default position by the "new" position resulting from the noise of the encoder values. After system got initialized and started i sent a command to the motor and noticed it is holding its position.
The reason for introducing a counter is only to manage the "setup current position as default position" so the default position gets set once and then is waiting" for me to send new positions from moveit2 or from joint_states_publisher_gui or even from the command line.
I know its not an elegant way to solve this issue but for my robot arm it works and i am waiting for the maintainers to come up with an alternative solution.

hope this helps and if you have further topics you would like to exchange maybe we can meet on discord

wjx1991, 2024-03-22:
> Hi wjx1991, I guess you ran into the same problem i had with the ethercat_driver_ros2. Motor is slowly moving out of its default position? As i investigated while debugging the ethercat_driver i [...]

Hi, jaronski, I use the ethercat_driver_ros2 to control my arm of 6 DOF.  The slow movement issue is solved by using your solution. Now I develop the Moveit2 package and control the arm to move.  New problems occur.
1. when I send the command by  joint_states_publisher_gui,  the arm can move slowly  and the magnitude is very small. 
2. when i start the system, the arm or the joints jitter.
3. the real attitude of the arm is not coincide with that in rviz in moveit2 
do you have these problems? 
4.  the  frequency in your code is 1khz,  when  i set it as 1khz error happens.  could you please, if possible, provide your code .

jaronski, 2024-05-08:
Hey @wjx1991 how can i find you on discord?
I might help you better over there

wjx1991, 2024-05-09:
Thanks. What is your discord username， i will add and contact you  on discord

---- Replied Message ----
| From | ***@***.***> |
| Date | 05/08/2024 21:16 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [ICube-Robotics/ethercat_driver_ros2] slowly moving after enabling (Issue #100) |

Hey @wjx1991 how can i find you on discord?
I might help you better over there

wjx1991, 2024-05-11:
Hi jaronski，my discord name is wjx16820138

---- Replied Message ----
| From | ***@***.***> |
| Date | 05/08/2024 21:16 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [ICube-Robotics/ethercat_driver_ros2] slowly moving after enabling (Issue #100) |

Hey @wjx1991 how can i find you on discord?
I might help you better over there