ICube-Robotics/ethercat_driver_ros2 #111: EtherCAT jitter and Synchronization Error

alejomancinelli, 2024-03-15:
Hi, I've been having issues with jitter in the EtherCAT communication. For example, here the new target positions, in red, should be all equal in lenght, when in reality this isn't happening. 

[image]

Also, this is my dmesg output (no correlation with the image above)

[image]

Sometimes I have a Datagram TIMEOUT! follow by a Datagram UNMATCHED!. In addition to this, from time to time I get this Synchronization Error, that disables my motor, which automatically reconnects after a little while. 

I've read issue #101 in detail, which contains some of the same errors. I am currently working in Ubuntu with the preempt_rt patch activated, in a Virtual Machine (maybe this is algo causing some issues). I follow some of the advice but I got no results.  

Do you know any improvements I can make to my system configuration? I think my problem resides there. Any test I can make to help me locate the root of the problem are also welcome.

PieterjanCottignies, 2024-03-20:
Hi @alejomancinelli I'm still also debugging some of the same errors where this causes a "modem like noise" in my drives.

I came across the following topic on the igh Ethercat master repo: [Issue73](https://gitlab.com/etherlab.org/ethercat/-/issues/73)
I still need to try and get the dedicated device driver going, this seemed to solve a lot of datagram timeouts for the user mentioning the problem.

I'm also checking the DC-Sync mechanism to see if this could have something to do with the jitter but no luck on that part so far.

Please let me know if you find something.

hanafym, 2024-03-22:
Hi @alejomancinelli , based on my experience, using a VM deteriorates performance of ethercat significantly.

hanafym, 2024-03-22:
Also may I ask what software do  you use to plot these results?