ICube-Robotics/ethercat_driver_ros2 #119: SDO not correctly set when using alias

JensVanhooydonck, 2024-04-27:
Due to the use of ecrt_master_sdo_download the alias is ignored (will always use 0).
Therefor we should use the slave config and use ecrt_slave_config_sdo

https://gitlab.com/etherlab.org/ethercat/-/issues/110

yguel, 2024-04-29:
Thanks for the report, I will fix the code according to your suggestion.

JensVanhooydonck, 2024-05-04:
I have a working example on my branch.. But hard to create a PR because I currently have a lot of changes on the same branch. In the commit are also auto-formatting changes: 
https://github.com/JensVanhooydonck/ethercat_driver_ros2/commit/a75b5cf358afb9c2a5b63de48ecf4e358bf72c1e

I did add a few things:
 - Added Alias And Position to the slave config object (so we can match aftwerwards when setting the SDO) [Slave Config](https://github.com/JensVanhooydonck/ethercat_driver_ros2/commit/a75b5cf358afb9c2a5b63de48ecf4e358bf72c1e#diff-9d9c6250c93902d8d52a9ad4893240bed88a18c4e1a87e4c84b10796a843c240R163) [Creation of Slave Config](https://github.com/JensVanhooydonck/ethercat_driver_ros2/commit/a75b5cf358afb9c2a5b63de48ecf4e358bf72c1e#diff-121641bbfdd534089317794a716a789edb2a1c7b2eab62d46fdc53a6d3b7c06aR75)
 - Altered the [configSlaveSdo](https://github.com/JensVanhooydonck/ethercat_driver_ros2/commit/a75b5cf358afb9c2a5b63de48ecf4e358bf72c1e#diff-121641bbfdd534089317794a716a789edb2a1c7b2eab62d46fdc53a6d3b7c06aR133)