ICube-Robotics/ethercat_driver_ros2 #126: Undefined reference errors for yaml-cpp when building for jazzy, noble

robin-zealrobotics, 2024-06-04:
I tried building this package for jazzy, on ubuntu noble.
The yaml-cpp library results in many errors like these
```
1.308 /usr/bin/ld: CMakeFiles/test_ec_pdo_channel_manager.dir/test/test_ec_pdo_channel_manager.cpp.o: in function `ethercat_interface::EcPdoChannelManager::load_from_config(YAML::Node) [clone .isra.0] [clone .cold]':
1.308 test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x345): undefined reference to `YAML::InvalidNode::~InvalidNode()'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x34c): undefined reference to `typeinfo for YAML::InvalidNode'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x414): undefined reference to `YAML::InvalidNode::~InvalidNode()'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x41b): undefined reference to `typeinfo for YAML::InvalidNode'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x69e): undefined reference to `YAML::InvalidNode::~InvalidNode()'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x6a5): undefined reference to `typeinfo for YAML::InvalidNode'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x6e5): undefined reference to `YAML::InvalidNode::~InvalidNode()'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x6ec): undefined reference to `typeinfo for YAML::InvalidNode'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x780): undefined reference to `YAML::InvalidNode::~InvalidNode()'
1.308 /usr/bin/ld: test_ec_pdo_channel_manager.cpp:(.text.unlikely+0x787): undefined reference to `typeinfo for YAML::InvalidNode'
1.308 /usr/bin/ld: CMakeFiles/test_ec_pdo_channel_manager.dir/test/test_ec_pdo_channel_manager.cpp.o: in function `YAML::Node::Node<char const*>(char const* const&)':
1.308 test_ec_pdo_channel_manager.cpp:(.text._ZN4YAML4NodeC2IPKcEERKT_[_ZN4YAML4NodeC5IPKcEERKT_]+0xe5): undefined reference to `YAML::detail::memory::create_node()'
```

Might have something to do with jazzy having yaml_cpp_vendor at version 9.0.0 now?
On iron, jammy, this builds perfectly well.

yguel, 2024-06-04:
Thanks for reporting.
We have conducted no tests on Jazzy for the moment.
You are right, it certainly is related to packages having been updated, especially yaml_cpp_vendor.
I will put resources to rapidly come up with a solution.

yguel, 2024-06-05:
I have succeeded compiling the project by manually adding yaml-cpp as target dependencies for linking (see https://github.com/yguel/ethercat_driver_ros2.git branch jazzy).

For the moment, the tests do not pass, which is quite normal since ros2 control evolved a lot from humble.

My way of making the project compile is not orthodox and I am receiving help from clalancette (see https://github.com/ros2/yaml_cpp_vendor/issues/51).
In the mean time, I would advance the work on the tests to make ethercat_driver_ros2 available asap to the community of jazzy users.

yguel, 2024-06-06:
Hi robin-zealrobotics,
I successfully build and test the library against jazzy with a ubuntu noble docker container.
I pushed a new branch named jazzy.

Can you confirm that it works for you ?

robin-zealrobotics, 2024-06-07:
Yes, it builds perfectly now, thanks!
I'll see if I can test it on real hardware next week, depending on whether other packages prove hard to work for jazzy.

closed 2024-06-07 by yguel