ICube-Robotics/ethercat_driver_ros2 #129: Ethercat Master 1.6 and network device compatibility

AltinayGrass, 2024-06-10:
Hello, 
I would like share some information, I tested with igh master stable-1.6 it is OK. Also I would like to learn device compatibility on docker solution. Is it possible use igc network device rather than generic? If yes, should we configure with --enable-igc option? 

Here is some part of the Dockerfile

`RUN mkdir ec_dev && cd ec_dev && \

    git clone https://gitlab.com/etherlab.org/ethercat.git && \

    cd ethercat && \

    git checkout stable-1.5 && \

    ./bootstrap && \

    ./configure --prefix=/usr/local/etherlab --disable-kernel && \

    make && make install`

yguel, 2024-06-13:
Hi,
Thanks for reporting the tests with stable-1.6. 
I did not test the igc network device. As soon as I can test it, I will post report here.

wei1224hf, 2024-06-23:
I am using ethercat 1.6.0 , with intel net card. not using generic, using e1000e, 
found that I have to reaptly restart the ethercat I the slaves are poweroff.
planning to switch to 1.5.2

maplezc9399, 2024-07-10:
Hi, I'm using ethercat 1.6 with I225 network card.
I'm using Ubuntu 20.04 + 5.15.44 + Xenomai 3.2.1 (dovetail 5.15.44-dovetail3) + IgH stable - 1.6 version.
I can run the ethercat only with ec_generic, I failed with ec_igc after multiple attempts. Wondering if I can get any help here?

Really appreciate!

allzergdead, 2024-09-11:
I have added support for real-time network card drivers e100e, r8169, igc, and igb for the Linux 6.5.2 kernel in version 1.6.2. If you're willing to upgrade your kernel to 6.5.2, you can use them directly. The r8169 driver has some bugs, such as not being able to reconnect after the network cable is disconnected, but no other issues have been found so far. You can access it here: https://github.com/allzergdead/ethercat-master-linux6.5.2-

Additionally, I will be open-sourcing a motion control function library based on IGH, which will implement S7 positioning, velocity mode, torque mode control, and corresponding speed-tension and torque-tension control algorithm application code.

allzergdead, 2024-09-11:
Based on my tests, the EOE feature may have bugs in some brands of servo drives. I recommend disabling the EOE feature during compilation