ICube-Robotics/ethercat_driver_ros2 #136: Error Messages reading SDO Etherlab-EthercatMaster1.6

NilsLo, 2024-07-12:
By Reading SDOs 3 Errors occure:
1. Wrong IOCTL-VERSION_MAGIC
It is required to Set IOCTL_VERSION_MAGIC to 37 insted 31 in ec_master_async_io.hpp
2. SDO-Read: **Operation not Permitted**
In examples of IGH 
`master.open(EcMasterAsync::ReadWrite);` is used in upload function. 
3. Reading data as string:  **Failed to upload SDO: No buffer space available** 
IGH uses
```
  if (data_type->byteSize){
    data.target_size = data_type->byteSize;
  } else{
    data.target_size = DefaultBufferSize;
  }
  data.target = new uint8_t[data.target_size + 1];

```
in upload function file ethercat_sdo_srv_server.cpp

with default buffer size:
```
const int  DefaultBufferSize= 64 * 1024;
```
defined in data_convertion_tools.hpp

ibrahimsel, 2024-07-16:
For 1. you could refer to #134

szliucm, 2024-11-08:
I've faced the same problem: SDO-Read: Operation not Permitted. Have you solved it?

NilsLo, 2024-11-13:
> I've faced the same problem: SDO-Read: Operation not Permitted. Have you solved it?
Yes, I could solve it.
I changed in: 
ethercat_sdo_srv_server.cpp

void upload(
...
-    master.open(EcMasterAsync::Read);
+    master.open(EcMasterAsync::ReadWrite);
...

NilsLo, 2024-11-13:
> I've faced the same problem: SDO-Read: Operation not Permitted. Have you solved it?

Yes, I could solve it.
I changed in:
ethercat_sdo_srv_server.cpp

```
void upload(
...
-master.open(EcMasterAsync::Read);
+master.open(EcMasterAsync::ReadWrite);
...

```

szliucm, 2024-11-15:
Thanks a lot for your solution.