ICube-Robotics/ethercat_driver_ros2 #138: Ethercat behavior is unstable

GiladRobee, 2024-07-16:
We have encountered an error in our library usage and would appreciate some feedback on how to solve it.

we have 16 EtherCAT units (7 drives and 9  I/O modules)
we launch our application in  mode of operation 9 (velocity)
and in doing so when we launch one of the motors starts rotating in a random (or so it seems) direction., before we even give it a command.
closing the EtherCAT master and restarting it didn't help but restarting the machine (jetson-orin) did work

JensVanhooydonck, 2024-07-19:
Seems related to #125 ... Are u using CSP mode?

GiladRobee, 2024-07-21:
I have seen this issue, the problem is that we are using CSV,  when I override the default velocity to 0 I cannot override it later and the motors stays put.

ammaraljodah, 2024-09-12:
You need to ensure the controller that is claiming the velocity command interface sends zero at the start.
First ensure your motors are disabled.
Delay the launch of controllers after the state broadcaster has been launched.
When enabling the motors, send a message to the velocity controller to command all joints to zero velocity,  then enable the motors.