ICube-Robotics/ethercat_driver_ros2 #143: can not activate 2 masters!

ammaraljodah, 2024-08-09:
Hi, 
It looks like 
<param name="master_id">1</param>
is never been checked in the driver code and it always set itself to 0 no mater what value is set in this tag. 
is there a way to have two masters using this library? 
this is my xacro file 

  <xacro:macro name="schunk_egu80ec_gripper">

    <ros2_control name="schunk_egu80ec_gripper_system" type="system">
      <hardware>
        <plugin>ethercat_driver/EthercatDriver</plugin>
        <param name="master_id">1</param>
        <param name="control_frequency">100</param>
      </hardware>
..
...

yguel, 2024-08-09:
Hi, ammaraljodah
This is a feature that is not supported for the moment.
The value is never used and 0 is always passed. 
The feature is planned, hence the parameter, but the is not yet implemented, probably because no need arised.

Can you describe your setup, since it may be easy to fix it in your case.

- all the best,

ammaraljodah, 2024-08-09:
I have 7 servo drives talks to the master 0 and 3 gripper talks to master 1. I just want less load on the servo master. 
because i keep getting these errors 
`[ 2706.598190] EtherCAT 0: Domain 0: Working counter changed to 0/21.
[ 2706.871102] EtherCAT WARNING 0: 3 datagrams UNMATCHED!
[ 2707.577194] EtherCAT WARNING: Datagram 00000000e0db0d0c (domain0-0-main) was SKIPPED 1 time.
[ 2707.607210] EtherCAT 0: Domain 0: Working counter changed to 21/21.
[ 2713.689730] perf: interrupt took too long (2514 > 2500), lowering kernel.perf_event_max_sample_rate to 79500
[ 2714.387988] EtherCAT 0: Domain 0: Working counter changed to 0/21.
[ 2714.647140] EtherCAT WARNING: Datagram 00000000e0db0d0c (domain0-0-main) was SKIPPED 2 times.
[ 2714.871046] EtherCAT WARNING 0: 6 datagrams UNMATCHED!
[ 2715.397138] EtherCAT 0: Domain 0: 3 working counter changes - now 21/21.
[ 2722.389385] EtherCAT 0: Domain 0: Working counter changed to 0/21.
[ 2722.727090] EtherCAT WARNING: Datagram 00000000e0db0d0c (domain0-0-main) was SKIPPED 2 times.
[ 2722.870985] EtherCAT WARNING 0: 6 datagrams UNMATCHED!
[ 2723.397083] EtherCAT 0: Domain 0: 3 working counter changes - now 21/21.
[ 2730.388113] EtherCAT 0: Domain 0: Working counter changed to 0/21.
[ 2730.807030] EtherCAT WARNING: Datagram 00000000e0db0d0c (domain0-0-main) was SKIPPED 2 times.
[ 2730.871932] EtherCAT WARNING 0: 6 datagrams UNMATCHED!
[ 2731.397025] EtherCAT 0: Domain 0: 3 working counter changes - now 21/21.
[ 2738.388897] EtherCAT 0: Domain 0: Working counter changed to 0/21.
[ 2738.870876] EtherCAT WARNING 0: 6 datagrams UNMATCHED!
[ 2738.886982] EtherCAT WARNING: Datagram 00000000e0db0d0c (domain0-0-main) was SKIPPED 2 times.
`

yguel, 2024-08-20:
I have made a branch that will be able to make the parameter master be taken into account correctly: https://github.com/ICube-Robotics/ethercat_driver_ros2/tree/yguel-use-master-id
If you try it, could-you give us some feedback ?

kouge0510, 2025-04-16:
> I have made a branch that will be able to make the parameter master be taken into account correctly: https://github.com/ICube-Robotics/ethercat_driver_ros2/tree/yguel-use-master-id If you try it, could-you give us some feedback ?

hi,I want to use two masters in two urdf file, so do I need to add both urdf with your package in one launch file?