ICube-Robotics/ethercat_driver_ros2 #148: ec_master no longer working after kernel update

bryangd34, 2024-10-18:
I recently did a `sudo apt update` and `sudo apt upgrade` and my kernel got updated:

`uname -a`:
Linux philip-System-Product-Name 6.8.0-47-generic #47~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Oct  2 16:16:55 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

and now:
`sudo /etc/init.d/ethercat start`:
Starting EtherCAT master 1.6.1 modprobe: FATAL: Module ec_master not found in directory /lib/modules/6.8.0-47-generic
 failed
 
 I followed the instructions in the install guide for what to do after a Kernel update, but that didn't help. I also tried installing again completely from scratch after deleting the ethercat folder and rebooting. Still no luck.
 
 Does anyone have any advice please? Thanks.

yguel, 2024-10-18:
Hi, 
It is normal that after a kernel update the ethercat module stops working, as it is compiled for a certain kernel version.
Anyway it is very strange that that trying to install from scratch does not work.
The error you encountered states that the module is not where it is supposed to be, hence its install failed.

Could you output the result from those commands (go in the folder where you have the source of ethercat):

1)
 ```bash
./bootstrap
```
2)
```bash
./configure --prefix=/usr/local/etherlab  --disable-8139too --disable-eoe --enable-generic```
```
3)
```bash
 make all modules
```
4)
```bash
 sudo make modules_install install
```
This one should install the module in the kernel folder

5) 
```bash
sudo depmod
```

I am working on a debian package to help install (It is not ready for distribution, but you can test it, it may help)
the project source and documentation is here: https://github.com/ICube-Robotics/ethercat_igh_dkms
And you can download the debian package here: https://seafile.unistra.fr/f/7b38a17736d14a08a8ee/?dl=1

bryangd34, 2024-10-19:
Hi @yguel,

Thanks a lot. After having a closer look at the outputs of those, I found the issue. Specifically `make all modules` :
/bin/sh: 1: gcc-12: not found

I then updated gcc to gcc-12 and the issue was solved:

reference: [reference](https://askubuntu.com/questions/1500017/ubuntu-22-04-default-gcc-version-does-not-match-version-that-built-latest-defaul)

closed 2024-10-19 by bryangd34