ICube-Robotics/ethercat_driver_ros2 #151: [EcCiA402Drive] Fault with status word: 4744 directly after Operation Enabled, but not every time

bryangd34, 2024-12-10:
I am working with a [ZeroErr drive unit](https://github.com/ZeroErrControl/eRob_ROS2_MoveIt) and am using the generic plugin EcCiA402Drive.

I would say approximately 90% of the time, when I try launch and bring up the drive, I get the following:
...
[ros2_control_node-1] Slave: State 0x01.
[ros2_control_node-1] Slave: State 0x02.
[ros2_control_node-1] Domain: WC 3.
[ros2_control_node-1] Domain: State 2.
[ros2_control_node-1] STATE: Fault with status word :648
[ros2_control_node-1] Slave: State 0x04.
[ros2_control_node-1] Master AL states: 0x0A.
[ros2_control_node-1] Slave: State 0x08.
[ros2_control_node-1] Slave: operational.
[ros2_control_node-1] STATE: Switch on Disabled with status word :4816
[ros2_control_node-1] STATE: Ready to Switch On with status word :4785
[ros2_control_node-1] STATE: Switch On with status word :4787
[ros2_control_node-1] STATE: Operation Enabled with status word :4791
[ros2_control_node-1] STATE: Fault with status word :4744
[ros2_control_node-1] STATE: Switch on Disabled with status word :4816
[ros2_control_node-1] STATE: Ready to Switch On with status word :4785
[ros2_control_node-1] STATE: Switch On with status word :4787
[ros2_control_node-1] STATE: Operation Enabled with status word :4791
[ros2_control_node-1] STATE: Fault with status word :4744
...

and then about 10% of the time, the drive is successfully brought up:
[ros2_control_node-1] Slave: State 0x01.
[ros2_control_node-1] Slave: State 0x02.
[ros2_control_node-1] Domain: WC 3.
[ros2_control_node-1] Domain: State 2.
[ros2_control_node-1] Slave: State 0x04.
[ros2_control_node-1] STATE: Ready to Switch On with status word :689
[ros2_control_node-1] STATE: Switch on Disabled with status word :4816
[ros2_control_node-1] Slave: State 0x08.
[ros2_control_node-1] Slave: operational.
[ros2_control_node-1] STATE: Ready to Switch On with status word :4785
[ros2_control_node-1] STATE: Switch On with status word :4787
[ros2_control_node-1] STATE: Operation Enabled with status word :4791

From here, I can control the drive successfully. 

Does anyone know how I may get the drive to consistently start successfully?

See also this [issue](https://github.com/ICube-Robotics/ethercat_driver_ros2/issues/87)

pzanne, 2024-12-10:
Hello @bryangd34,

To help us to debug your problem, could you share your config file? 
In your first case, the driver seems in error before start-up:
```
[ros2_control_node-1] Domain: State 2.
[ros2_control_node-1] STATE: Fault with status word :648
[ros2_control_node-1] Slave: State 0x04.
```
And the error does not seem resettable, as you are probably in automatic reset mode the ethercat ros driver tries indefinitely to reset it.
Could you read the value in error register 0x603F ? with the command :

`ethercat -p XXX upload -t uint16 0X603F 0X0`

replace the `XXX` with the position on the ethercat bus of the driver in error .

Could you also share the ethercat kernel logs?
Before launching the ros node, activate the debug mode of ethercat igh stack :

`ethercat debug 1`

delete the kernel messages :

`sudo dmesg -c`

then launch the ros node and stop it after a few seconds (the time for an error reset cycle) and finally retrieve the kernel messages:
`sudo dmesg -c | grep EtherCAT >logzeroerr.txt `

Ideally, you should follow this procedure for a functional case and a non-functional case

bryangd34, 2024-12-11:
Thanks @pzanne for all your suggestions.
It is working now. 
I was about to upload my config file, when taking a look at it, I saw it was missing a target torque rpdo. I added it, out of the sake of completeness, not thinking it would fix the issue. But it seems like it did. 
Here is the config file anyway, in case you have some comments:

```
vendor_id: 0x5A65726F
product_id: 0x00029252
assign_activate: 0x0300 # DC Synch register
auto_fault_reset: true  # true = automatic fault reset, false = fault reset on rising edge command interface "reset_fault"
auto_enable_set: true
dc_mode: true

#sdo:
#  - {index: 0x4602, sub_index: 0, type: uint32, value: 1}

rpdo:  # RxPDO = receive PDO Mapping
  - index: 0x1600
    channels:
      - {index: 0x607a, sub_index: 0, type: int32, command_interface: position, factor: 1000, default: .nan}  # Target position
      - {index: 0x60fe, sub_index: 0, type: int32, default: 0}  # Digital Output Functionalities
      - {index: 0x6040, sub_index: 0, type: uint16, default: 0}  # Control word
      - {index: 0x60ff, sub_index: 0, type: int32, command_interface: velocity, default: 0}
      - {index: 0x6071, sub_index: 0, type: int16, default: 0}  # Target torque

tpdo:  # TxPDO = transmit PDO Mapping
  - index: 0x1A00
    channels:
      - {index: 0x6064, sub_index: 0, type: int32, state_interface: position, factor: 0.001}  # Position actual value
      - {index: 0x606c, sub_index: 0, type: int32, state_interface: velocity}
      - {index: 0x6077, sub_index: 0, type: int16, state_interface: effort}
      - {index: 0x60fd, sub_index: 0, type: int32}  # Digital Input Functionalities
      - {index: 0x6041, sub_index: 0, type: uint16}  # Status word
      - {index: 0x603f, sub_index: 0, type: uint16} # ERROR CODE
      - {index: 0x6061, sub_index: 0, type: uint16, default: 8}

sm:  # Sync Manager
  - {index: 0, type: output, pdo: ~, watchdog: disable}
  - {index: 1, type: input, pdo: ~, watchdog: disable}
  - {index: 2, type: output, pdo: rpdo, watchdog: enable}
  - {index: 3, type: input, pdo: tpdo, watchdog: disable}
```

FYI, the drive seems to be in error most times at start up, even when successfully entering operation enabled and remaining there:
[ros2_control_node-1] Slave: State 0x01.
[ros2_control_node-1] Slave: State 0x02.
[ros2_control_node-1] Domain: WC 3.
[ros2_control_node-1] Domain: State 2.
[ros2_control_node-1] STATE: Fault with status word :648
[ros2_control_node-1] Slave: State 0x04.
[ros2_control_node-1] Slave: State 0x08.
[ros2_control_node-1] Slave: operational.
[ros2_control_node-1] STATE: Switch on Disabled with status word :720
[ros2_control_node-1] STATE: Ready to Switch On with status word :4785
[ros2_control_node-1] STATE: Switch On with status word :691
[ros2_control_node-1] Master AL states: 0x0A.
[ros2_control_node-1] STATE: Operation Enabled with status word :4791

ZeroErrControl, 2024-12-26:
> I am working with a [ZeroErr drive unit](https://github.com/ZeroErrControl/eRob_ROS2_MoveIt) and am using the generic plugin EcCiA402Drive.
>
> I would say approximately 90% of the time, when I try [...]

I attempted to address this issue by modifying the underlying code of ethercat_driver_ros2. The results of these attempts are detailed in the [fhttps://github.com/ICube-Robotics/ethercat_driver_ros2/issues/154] (https://github.com/ICube-Robotics/ethercat_driver_ros2/issues/154).

ZeroErrControl, 2024-12-27:
Some users reported issues where the ethercat_ros2 interface could not bring eRob into the OP (Operational) state. To address this, I have provided a solution in this project. Please refer to the detailed steps here: [Solution Link](https://github.com/ZeroErrControl/eRob_ROS2_MoveIt).

I have tested this solution on a six-axis robotic arm, and the results are as follows:

Successfully entered DC mode: The EtherCAT Distributed Clock (DC) is functioning correctly.
Successfully entered OP state: All EtherCAT slave devices transitioned to the operational state, and the robotic arm operates as expected.
Feel free to try it out and share your feedback! If you encounter any issues during implementation, please open an issue here, and I’ll assist you as soon as possible.

gordon-dolphinlabs, 2025-06-18:
Hi -- I am curious: does `dc_mode: true` in the config file actually do anything? I can't find a spot where this parameter is loaded in the codebase and am currently debugging some dc-related issues, so this piqued my interest. Thanks