ICube-Robotics/ethercat_driver_ros2 #152: using a single master but launching multiple devices separately

aatb-ch, 2024-12-15:
Hello there, thanks for the awesome work!

I have a project where I'm running a single master, but I would like to launch separately different devices (IO module, drives, etc) depending on the situation, and would like to keep the config and launch files of these devices separate.

I've already gotten working an EL2008 and EL7031 modules, each work and launch fine alone, and i'd like to keep their config as separate packages, and launching them independently (and sometimes not launching one at all). Is this possible with a shared master without writing a single ros2_control configuration? Launching each throw the expected `Failed to reserve master: Device or resource busy` on the second one.

Is there an easy way I'm missing? or the whole bus configuration has to be declared under a single ros2_control config?

yguel, 2024-12-15:
Hi, 
Thanks for your message. I would like to know a bit more about your need to understand correctly, especially because with ethercat and the current setting, the communication frame is fixed (we do not support for the moment hotplug, it is in our roadmap though).
What does it mean: all the memory in the frame is allocated and for all the slaves communicating on the bus, so you cannot add a slave without the master knowing how to communicate with it beforehand.
So basically it means that you have to make your bus act as if all the slaves communicate and just ignore the data when one or all the slaves are not communicating. So configure your system such that all slaves are declared to the master, and just ignore the slaves that are not relevant at a certain time.
Is-it a suitable solution for you ?

For the moment, we do not check that a slave is not communicating (but we consider it a number one feature to add : to have such a check as a security measure), so somehow it is a chance for your need.
By the way pzanne is currently looking at that feature, so your request arrive just on time, if we have to make the check optional such that your use case is taken into account.

Anyway, that is a very good moment to tell us the most you can about your use case.

aatb-ch, 2024-12-16:
Hi, thanks a lot for the reactivity!

So my use case is that we have developed a system for film production that orchestrates multiple type of devices, that are often used in different combination depending on the production requirements. Practically that means we might need one day a Universal Robot along digital outputs, another day an ABB robot along a servo for a linear axis, another day a stepper for a turntable, another day all together, and another day nothing but the ABB and Universal robots together. Currently this works by having independent devices on adhoc adapters, usb-serial, usb-rs-485, or raspberry pi shields for relay outputs, CAN adapters, etc. It's very easy to manage and maintain each devices as independent packages and launch them individually when needed one or the other.

For robustness and future-proofing the system we have decided to migrate everything to ethercat with the according Beckhoff modules, replacing the stack of raspi shields and dangling usb adapters with just a compute module running standard ethernet for networking and a second ethernet for ethercat. We are really happy with the direction things are shaping up, but I just wonder how best to keep the devices as separate packages that can be launched individually. I indeed could simply declare my entire configuration as a single ros2_control since i will not physically unplug the modules, they will always be on the bus, but for example if the servo motor is not plugged to the drive, I would prefer not to launch the servo driver at all, rather than being forced to launch it and having it error out.

I also could have a set of configuration files and pick the one fitting the combination required, but this would not really be nice to maintain when adding new devices. Would there be a way to launch the master (with the "full" frame including all possible devices on the bus) but launch separately the controllers for each devices? Sort of launching the hardware interface but separately claiming these when launching the controllers only?

I could imagine ideally having a package for the master itself, which could have all the hardware implementation of each modules, and then each "logical" device being a separate package and launching only the controllers claiming the interfaces declared on the master package? Does that make sense?

yguel, 2024-12-16:
Hi,

> Would there be a way to launch the master (with the "full" frame including all possible devices on the bus) but launch separately the controllers for each devices? Sort of launching the hardware interface but separately claiming these when launching the controllers only?

That is exactly what the decoupling between hardware interfaces and controllers is useful for. So yes, you could exactly do that.

> I could imagine ideally having a package for the master itself, which could have all the hardware implementation of each modules, and then each "logical" device being a separate package and launching only the controllers claiming the interfaces declared on the master package? Does that make sense?

Totally, that is what I would aim for. 

Keep in mind, that it would be always possible (with only minor changes to your launch files) to have a set of configurations (UR+ABB+turntable, UR+ABB, ABB+turntable, ABB+servo, etc.) and launch the hardware interface (aka the master) with the appropriate configuration, if you need that later.
But if all the Beckhoff modules are to stay, it is more simple to have one big configuration with all hardware interfaces declared.

By the way thanks for sharing your use case. We will take it into account to design our security layer, to make it a configuration option to not raise an error when hardware is not communicating.

aatb-ch, 2024-12-16:
Alright, I see the various options for now then, thanks a lot for the clarifications!

I'll experiment with the options and see what fits best currently, trying mostly to avoid having to maintain an exhaustive combination of templates, I might be able to get away with conditions in a single launch file hopefully, will report back if i'm stuck. thanks again!

closed 2024-12-16 by aatb-ch

yguel, 2024-12-16:
You have to be careful about one thing though: if all your systems must be powered and able to communicate.
The reason, is that if some slaves are daisy chained, and if you want to power only parts of some slaves on a daisy chain, you will cut the communication line, because the unpowered slaves will cut the line.
There is a potential solution with a star topology, with all the slaves in the same configuration subset being in the same branch, but in this case, the solution will use aliases (untested/partially implemented feature of ethercat_driver_ros2 package).
It is part of our roadmap to support it, but the devs are not scheduled yet.

Will it be a problem to have all slaves powered and able to communicate, even if not in use ?

aatb-ch, 2024-12-16:
I dont believe it is a problem, the goal being to have just a single chain of modules on a single coupler and not touching it. The issue lies more in the peripherals wired to these modules, but is mostly a problem for steppers or servos, if disconnected the drives would just fault if part of the config file. The rest of the modules (ios, rs-485, etc) are not really a concern if always-on.

The main concern really was if it's possible to separately launch/stop devices on the same master but I think for now it's clear that a master should be configured with a full list on launch, and I can optionally launch controllers for each devices when needed, that's a totally acceptable solution and worst case I will fall back to 2 launch file, with or without the modules that wont be happy if the external hardware is not wired.