ICube-Robotics/ethercat_driver_ros2 #154: Cannot Enable eRob [ZeroERR CIA402 Motor] DC Mode and Random OP State Entry with `ethercat_driver_ros2`

ZeroErrControl, 2024-12-26:
## **Description**

## Environment Details
- OS: Ubuntu 20.04
- ROS 2 Version: Humble
- Motor: ZeroERR eRob (CIA402 protocol compliant)
- Driver: ethercat_driver_ros2
- Master: IGH EtherCAT Master (for comparison)

I am using `ethercat_driver_ros2` to integrate ZeroERR's **eRob servo motor** (compliant with the CIA402 protocol) with MoveIt for motion control. However, the following issues were encountered:

1. **eRob cannot reliably enter Distributed Clock (DC) mode.**
2. **The slave shows random behavior when entering OP state,** often remaining in **Safe OP + Error** instead of OP state.

These issues prevent the seamless use of `ethercat_driver_ros2` to control the eRob motor.

---

## **Experiments and Observations**

### **Experiment 1: Removing DC Configuration Code**
I removed the following code block from `ec_master.cpp`:
```cpp
if (slave->assign_activate_dc_sync()) {
    struct timespec t;
    clock_gettime(CLOCK_MONOTONIC, &t);
    ecrt_master_application_time(master_, EC_NEWTIMEVAL2NANO(t));
    ecrt_slave_config_dc(
      slave_info.config,
      slave->assign_activate_dc_sync(),
      interval_,
      interval_ - (t.tv_nsec % (interval_)),
      0,
      0);
}
```
### Results:
eRob successfully entered OP state.
Running `ethercat upload -p0 0x1c32 1` returned:
``` bash
0x0001 1
```
This indicates that eRob remained in Sync Manager (SM) mode rather than DC mode.
Conclusion:
Removing the DC configuration allows eRob to enter OP state but prevents DC mode activation.

## Experiment 2: Forcing DC Configuration
I removed the condition `ethercat_driver_ros2/ethercat_interface/src/ec_master.cpp` check and forced the DC configuration code to execute:
```bash

struct timespec t;
clock_gettime(CLOCK_MONOTONIC, &t);
ecrt_master_application_time(master_, EC_NEWTIMEVAL2NANO(t));
ecrt_slave_config_dc(
  slave_info.config,
  slave->assign_activate_dc_sync(),
  interval_,
  interval_ - (t.tv_nsec % (interval_)),
  0,
  0);
```
### Results:
eRob successfully entered DC mode.
Running `ethercat upload -p0 0x1c32 1` returned: `0x0002 2`

This confirms that eRob was in DC mode.
However, eRob failed to enter OP state and remained in Safe OP + Error state.
Kernel logs showed the following messages:

```bash
[76477.776110] EtherCAT 0: Starting EtherCAT-OP thread.
[76478.371933] EtherCAT WARNING 0: 3 datagrams TIMED OUT!
[76478.371944] EtherCAT WARNING 0: 4 datagrams UNMATCHED!
[76484.748536] EtherCAT WARNING 0-0: Slave did not sync after 5000 ms.
[76494.198581] EtherCAT ERROR 0-0: Failed to set OP state, slave refused state change (SAFEOP + ERROR).
[76494.208694] EtherCAT ERROR 0-0: AL status message 0x001A: "Synchronization error".
```
### Conclusion:
Forcing DC configuration enables DC mode but causes synchronization errors, preventing OP state entry.

## Experiment 3: Using IGH EtherCAT Master
I used a custom IGH EtherCAT master implementation ([code link](https://github.com/ZeroErrControl/eRob_IGH_EtherCAT/blob/main/src/igh_driver.cpp) )to control eRob.

### Results:
Successfully enabled DC mode.
eRob entered OP state without errors.
All functionalities worked as expected.
### Conclusion:
Using the IGH EtherCAT master confirms that the issue is specific to ethercat_driver_ros2.

## Analysis
DC Configuration Implementation:

- The interval_ parameter and the shift_time calculation in ecrt_slave_config_dc might be incorrect, leading to timing mismatches.
- The master application time (ecrt_master_application_time) may not align with the slave's reference clock, causing synchronization issues.
- The DC configuration might fail to fully activate all required SYNC managers (e.g., SYNC0).
Synchronization Errors:

- Kernel logs show AL status message 0x001A ("Synchronization error"), indicating that the master and slave clocks are not properly synchronized.
Contrast with IGH EtherCAT Master:

- The same hardware and configuration worked perfectly with the IGH EtherCAT master, suggesting that ethercat_driver_ros2 has issues in handling DC mode and synchronization.

## Recommendations
Verify and Fix DC Configuration:

- Ensure interval_ and shift_time parameters are correctly calculated based on the desired cycle time and reference clock.
Check that ecrt_master_application_time properly aligns with the slave's reference clock.
Confirm that both SYNC0 and SYNC1 are properly configured.
Improve Synchronization Logic:

- Add debug logs to verify the execution and effects of ecrt_master_sync_reference_clock and ecrt_master_sync_slave_clocks.
Validate that the master application time is set before attempting synchronization.
Enhance Compatibility:

- Review and ensure ethercat_driver_ros2 fully supports CIA402-compliant devices like eRob.
Add configuration options or examples specifically tailored for CIA402 motors.

ZeroErrControl, 2024-12-27:
I have preliminarily resolved the issue where the ethercat_ros2 interface fails to bring eRob into the OP (Operational) state. The detailed solution is provided in the following project: [Project Link](https://github.com/ZeroErrControl/eRob_ROS2_MoveIt).

pzanne, 2025-01-06:
Hello ZeroErrControl,
Thank you for your feedback,

Improving the behaviour of the distributed clock is in our development plan, and we could use your use case to make progress on this point. The cia402 drivers that we've been able to test (technosoft, elmo, synapticon, maxon) so far haven't any problems for switching to operational mode in DC mode. Would it be possible to access your servomotor for testing purposes (remotely, for example)? 

Your kernel logs in experiment 2 show ethercat frame losses (DATAGRAM timed out and DATAGRAM unmatched). Could you test the branch `humble-issue-147` and tell me if this changes anything when switching to operational mode? 

Could you also send me your kernel logs for experiment 3? 
Best 
Philippe

ZeroErrControl, 2025-01-08:
Dear Philippe,

Thank you very much for your detailed response and for seriously considering my suggestions.

Based on your feedback, I have formulated the following experimental plan:

Comparison of DC Performance:
We will conduct a comparison experiment between the CIA402 drivers you mentioned (Technosoft, Elmo, Synapticon, Maxon) and our eRob (ZeroErr CIA402 Motor) under Distributed Clock (DC) mode.
Additionally, the idea of remote access to my servo motor is certainly feasible. Could you provide guidance on how to proceed with the connection and continuous operation?

Testing the Humble-issue-147 Branch:
I will test the branch you mentioned and carefully observe and record any changes during the switch to operational mode.

Kernel Logs for Experiment 3:
I will repeat Experiment 3 and organize the kernel logs to share them with you for further analysis.

These experiments will require some time to complete, but I will ensure the results are well-organized and sent to you as soon as possible.

Thank you again for your support, and I look forward to continuing this collaboration!

Best regards,
ZeroErrControl

pzanne, 2025-01-20:
Hi @ZeroErrControl 

We've carried out some tests, and it's possible that we've found the source of the clock synchronisation problem. 
It is possible that the problem is caused by the cycle time management in ros2_control. If a cycle time is too long in relation to the period, the read/update/write events are buffered in ros2_control. As soon as ros2_control takes over, it tries to empty this buffer, leading to temporal instability when calling the read/write functions of our ethercat driver (very bad for the DC synchronization).
We are currently looking for a solution to this problem. As soon as we have something working we'll get back to you.
Best
Philippe

ZeroErrControl, 2025-01-21:
Dear Philippe,

Your identified issue indeed matches the problem we've observed. When the DC mode fails to initialize, the slave falls back to SM mode, and it seems the slave's configuration gets reset. Thank you very much for pinpointing the issue, and we look forward to your solution.

Best regards,
ZeroErrControl

pzanne, 2025-03-27:
Sorry for taking so long to reply, I've been tied up with other projects.
I've made some progress on the synchronization problem, I have a corrected version of our driver available here : 
`https://git.unistra.fr/zanne.philippe/ethercat_driver_ros2`
under the safety-humble-dcsync branch. On the configuration I have for testing (3 technosoft ipos 3604 drivers) it works in the same way as going directly through the ethercat igh api. However, I've found that the ref to master synchronization method requires a frequency of at least 1000Hz. Below that, the success of the synchronization seems random. It also seems advisable to use a dedicated driver for the ethernet card (not the generic driver) and a preempt rt system.  I'm investigating this further to better understand the limits.
I've also tested your program available on your github (`https://github.com/ZeroErrControl/eRob_IGH_EtherCAT`), which I've adapted to my configuration, and found the same constraints and found the same constraints to work well (1000Hz and preempt rt recommended).
Would you like us to organize a test session on your setup?  Or can you give me some feedback?
best
Philippe

ZeroErrControl, 2025-04-16:
Sorry, I've also been tied up with other projects recently and haven't had the chance to test the solution you provided. I plan to run the tests by the end of this month and will share the results with you as soon as they are available. Thank you for your understanding and support!

ZeroErrControl, 2025-07-14:
Sorry, We couldn't find a complete testable project in the link you provided, so we were not able to give feedback. Would it be possible for you to upload the full project to facilitate testing?

we have recently been tied up with other projects and just got back to the main development line. We're currently planning to develop a host-side application based on the ros_ethercat_driver project, so we would like to check whether the relevant issues have been resolved.

ZeroErrControl, 2025-07-17:
Thank you very much for your help. The issue has been resolved.
I will document the detailed steps.

closed 2025-07-17 by ZeroErrControl

pzanne, 2025-07-17:
Hello,
Have you used our modifications or have you found your own solution? If the latter, I would appreciate hearing your ideas.
Best
Philippe

ZeroErrControl, 2025-07-17:
Hello,

Regarding the previously reported issue where the eRob joint could not enter OP state, I have conducted further investigation and discovered that it was caused by a typo in my configuration file — I mistakenly wrote assign_activate as assign-activate, which prevented the slave device from being properly activated.

This typo may have originated from an earlier version of the official documentation I referred to. I recall attempting to modify this parameter correctly at some point, but due to the time that has passed, I can’t clearly remember the reasoning behind the current setting. As a result, the Distributed Clock (DC) mode was never correctly configured, leading to a series of seemingly complex issues.

It is now confirmed that the root cause of the problem was simply this misconfiguration on my side. I sincerely apologize for the confusion and the time this may have cost you, and I truly appreciate your continued support and understanding.