ICube-Robotics/ethercat_driver_ros2 #156: GenericEcSlave Handling RPOs assigned to Multiple Sync Managers

cgrigas, 2025-01-27:
I'm using a Beckhoff EP2339-0021 GPIO device which uses a non-standard SM configuration. 

It has 16x RPDOs, 8x of which are assigned to SM0 and 8x are assigned to SM1. If I'm understanding the GenericEcSlave implementation correctly, it can only handle a 1-1 mapping of RPDO or TPDO to SM, meaning that there is no support to assign different groups of RPDOs to separate SMs. I'm attaching a snippet of the relevant part of the device's ESI below

```
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1600</Index>
					<Name>Channel 17</Name>
					<Entry>
						<Index>#x7000</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1601</Index>
					<Name>Channel 18</Name>
					<Entry>
						<Index>#x7010</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1602</Index>
					<Name>Channel 19</Name>
					<Entry>
						<Index>#x7020</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1603</Index>
					<Name>Channel 20</Name>
					<Entry>
						<Index>#x7030</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1604</Index>
					<Name>Channel 21</Name>
					<Entry>
						<Index>#x7040</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1605</Index>
					<Name>Channel 22</Name>
					<Entry>
						<Index>#x7050</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1606</Index>
					<Name>Channel 23</Name>
					<Entry>
						<Index>#x7060</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1607</Index>
					<Name>Channel 24</Name>
					<Entry>
						<Index>#x7070</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x1608</Index>
					<Name>Channel 25</Name>
					<Entry>
						<Index>#x7080</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x1609</Index>
					<Name>Channel 26</Name>
					<Entry>
						<Index>#x7090</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x160a</Index>
					<Name>Channel 27</Name>
					<Entry>
						<Index>#x70a0</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x160b</Index>
					<Name>Channel 28</Name>
					<Entry>
						<Index>#x70b0</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x160c</Index>
					<Name>Channel 29</Name>
					<Entry>
						<Index>#x70c0</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x160d</Index>
					<Name>Channel 30</Name>
					<Entry>
						<Index>#x70d0</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x160e</Index>
					<Name>Channel 31</Name>
					<Entry>
						<Index>#x70e0</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
				<RxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x160f</Index>
					<Name>Channel 32</Name>
					<Entry>
						<Index>#x70f0</Index>
						<SubIndex>1</SubIndex>
						<BitLen>1</BitLen>
						<Name>Output</Name>
						<DataType>BOOL</DataType>
					</Entry>
				</RxPdo>
```

JensVanhooydonck, 2025-02-08:
I have solved this by splitting the config file in two files, one for SM0 and one for SM1, also setting the SM configs accordingly.

pzanne, 2025-02-18:
Hello,
we're aware of this limitation on sync managers, and it's part of our development roadmap to extend the configuration possibilities on the sync manager.
Best
Philippe