ICube-Robotics/ethercat_driver_ros2 #158: .nan Position Default Always Sending 0 to Copley NES Drive

cgrigas, 2025-02-12:
I'm using a Copley NES drive and trying to operate it in CSP mode. When I try to use the .nan default value however it always sends 0 as the position command. I've confirmed this by querying the 0x607a object with an SDO upload. The drive always rejects this change because it's too large of a step from its real position and it triggers a following error. 

If I set a value close to the real current position value in the default then the drive will properly control to that position on startup. Querying the 0x607a object here will give the value that I set in the default. So the drive is properly configured, going to the correct operation state, and the driver is sending the value that I've commanded. The failure seems to be in the .nan feedback interface. A potentially related issue is that I'm also not able to send a command to the position command interface with a ROS2 Control ForwardCommandController, again it's always 0.

I'm successfully using the .nan position feedback with other devices, but there seems to be something in particular wrong with this specific drive combination. I think it may be related to the drive using a TPDO/RPDO in a non standard address range (0x1700 for RPDO and 0x1B00 for TPDO), but that's an unconfirmed suspicion. 

I've spent a few days debugging this so any help would be greatly appreciated.

YAML config: 

```
vendor_id: 0x000000AB # Copley Controls vendor ID (171 in hex)
product_id: 0x00002080 # NES drive product code
revision_number: 0x00010003 # From ESI Type RevisionNo
auto_fault_reset: true
auto_enable_set: true

sdo:
  # PDO mapping configuration
  
  - { index: 0x1A00, sub_index: 0, type: uint8, value: 0 }
  - { index: 0x1A01, sub_index: 0, type: uint8, value: 0 }
  - { index: 0x1A02, sub_index: 0, type: uint8, value: 0 }
  - { index: 0x1A03, sub_index: 0, type: uint8, value: 0 }
  - { index: 0x1600, sub_index: 0, type: uint8, value: 0 }
  - { index: 0x1601, sub_index: 0, type: uint8, value: 0 }
  - { index: 0x1602, sub_index: 0, type: uint8, value: 0 }
  - { index: 0x1603, sub_index: 0, type: uint8, value: 0 }

  - { index: 0x1C12, sub_index: 0, type: uint8, value: 0 } # Number of assigned RxPDOs
  - { index: 0x1C12, sub_index: 1, type: uint16, value: 0x1700 }
  - { index: 0x1C12, sub_index: 0, type: uint8, value: 1 } # Number of assigned RxPDOs

  - { index: 0x1C13, sub_index: 0, type: uint8, value: 0 } # Number of assigned TxPDOs
  - { index: 0x1C13, sub_index: 1, type: uint16, value: 0x1B00 } # Assign TxPDO
  - { index: 0x1C13, sub_index: 0, type: uint8, value: 1 } # Number of assigned TxPDOs

  # Operation mode
  - { index: 0x6060, sub_index: 0, type: int8, value: 8 } # CSP Mode

rpdo:
  # POSITION CONTROL PDOS
  - index: 0x1700
    channels:
      - {
          index: 0x6040,
          sub_index: 0,
          type: uint16,
        }
      - {
          index: 0x607a,
          sub_index: 0,
          type: int32,
          command_interface: position,
          default: .nan,
          factor: 1000000,
          offset: 1211580,
        }
      - {
          index: 0x60b1,
          sub_index: 0,
          type: int32,
          factor: 10000000,
          offset: 0,
        }
      - {
          index: 0x60b2,
          sub_index: 0,
          type: int16,
          factor: 89.2857142857,
          offset: 0,
        }
tpdo:
  - index: 0x1B00
    channels:
      - {
          index: 0x6041,
          sub_index: 0,
          type: uint16,
        }
      - {
          index: 0x6064,
          sub_index: 0,
          type: int32,
          state_interface: position,
          factor: 0.000001,
          offset: -1.21158,
        }
      - { index: 0x60f4, sub_index: 0, type: int32, factor: 1.0, offset: 0 }
      - {
          index: 0x606c,
          sub_index: 0,
          type: int32,
          state_interface: velocity,
          factor: 0.000001,
          offset: -1.21158,
        }
      - {
          index: 0x6077,
          sub_index: 0,
          type: int16,
          state_interface: effort,
          factor: 0.0112,
          offset: 0,
        }
```

filippo-bosi, 2025-02-12:
Hello, I encountered a similar issue. In my case, the root cause was the `joint_trajectory_controller`.

The problem was that the controller was being spawned too early, causing the position command to default to zero. I resolved this by introducing a delay before spawning the controller, making sure that the system had enough time to initialize properly.

Here’s the code I used in my launch file:
```
controller_spawner = TimerAction(
    period=controller_spawner_timeout,  # Delay before spawning controllers
    actions=[
        Node(
            package="controller_manager",
            executable="spawner",
            arguments=["joint_trajectory_controller", "-c", "/controller_manager"],
            output="screen",
        )
    ]
)
```

cgrigas, 2025-02-12:
Thanks for the response @filippo-bosi , but in my case it looks like it was a similar issue to this other issue #125 . I didn't have object 0x6061 mode of operation display mapped and that apparently prevents all commands to the target position object. This seems like a pretty important dependency that should be mentioned in the documentation

pzanne, 2025-02-18:
Hello,

That right, the importance of register 0X6061 is perhaps not sufficiently highlighted in the documentation, and we are going to update the documentation in this respect. 
When you request a change of operating mode via register 0x6060, it may be that for some reason internal to the driver the change is not validated. To find out the current operating mode, you need to have the return of register 0X6061, and depending on the value in the register, it may or may not be possible to write a new reference.
if you include this register in your configuration, does that solve your problem? 
Best
Philippe

cgrigas, 2025-02-18:
@pzanne yes, including the 0x6061 register in my configuration does resolve the issue. I'm now able to use the .nan position feedback and command the position to the drive via the command interface. This 0x6061 requirement seems to only be required in CSP operation mode though, I was previously operating this device in CSV fine without having the 0x6061 feedback.

Thanks for agreeing to update the documentation

closed 2025-02-18 by cgrigas

kouge0510, 2025-04-16:
> [@pzanne](https://github.com/pzanne) yes, including the 0x6061 register in my configuration does resolve the issue. I'm now able to use the .nan position feedback and command the position to the [...]

```
# Configuration file for KDE 系列伺服驱动器
vendor_id: 0x00010203  # 已修改
product_id: 0x00000402 # 已修改
assign_activate: 0x0300  # DC Synch register，我们程序没有，但是文档存在
auto_fault_reset: false  # true = automatic fault reset, false = fault reset on rising edge command interface "reset_fault"，未修改
sdo:  # sdo data to be transferred at drive startup，未修改
   - {index: 0x6060, sub_index: 0, type: int8, value: 8} # Set interpolation time for cyclic modes to 10 ms，未修改

rpdo:  # RxPDO = receive PDO Mapping，未修改
  # - index: 0x1600
  #   channels:
  #     - {index: 0x6040, sub_index: 0, type: uint16, default: 0}  # Control word，已修改
  #     - {index: 0x607a,sub_index: 0,type: int32,command_interface: position,default: .nan}  # Target position，已修改
  #     - {index: 0x60ff, sub_index: 0, type: int32, default: 0}  # Target velocity，已修改
  #     - {index: 0x6071, sub_index: 0, type: int16, default: 0}  # Target torque
      # - {index: 0x6060, sub_index: 0, type: int8, default: 8}  # Mode of operation，已修改，文档里没有，但是代码有

  - index: 0x1601
    channels:
      - {index: 0x6040, sub_index: 0, type: uint16, default: 0}  # Control word，已修改
      - {index: 0x607a,sub_index: 0,type: int32,command_interface: position,default: .nan}  # Target position，已修改
      - {index: 0x60ff, sub_index: 0, type: int32, default: 0}  # Target velocity，已修改
      # - {index: 0x6071, sub_index: 0, type: int16, default: 0}  # Target torque
      - {index: 0x6060, sub_index: 0, type: int8, default: 8}  # Mode of operation，已修改，文档里没有，但是代码有
  # - index: 0x1602
  #   channels:
  #     - {index: 0x6040, sub_index: 0, type: uint16, default: 0}  # Control word，已修改
  #     - {index: 0x607a,sub_index: 0,type: int32,command_interface: position,default: .nan}  # Target position，已修改
  #     - {index: 0x60ff, sub_index: 0, type: int32, default: 0}  # Target velocity，已修改
  #     - {index: 0x6071, sub_index: 0, type: int16, default: 0}  # Target torque
  #     - {index: 0x60b8, sub_index: 0, type: uint16, default: 8}  # touch probe function，已修改，文档里没有，但是代码有
  # - index: 0x1603
  #   channels:
  #     - {index: 0x6040, sub_index: 0, type: uint16, default: 0}  # Control word，已修改
  #     - {index: 0x607a,sub_index: 0,type: int32,command_interface: position,default: .nan}  # Target position，已修改
  #     - {index: 0x60ff, sub_index: 0, type: int32, default: 0}  # Target velocity，已修改
  #     - {index: 0x6071, sub_index: 0, type: int16, default: 0}  # Target torque
  #     - {index: 0x60b8, sub_index: 0, type: uint16, default: 0}  # touch probe function，已修改，文档里没有，但是代码有
    
tpdo:  # TxPDO = transmit PDO Mapping
  # - index: 0x1a00
  #   channels:
  #     - {index: 0x6041, sub_index: 0, type: uint16}  # Status word，已修改
  #     - {index: 0x6064,sub_index: 0,type: int32,state_interface: position}  # Position actual value, 已修改
  #     - {index: 0x606c,sub_index: 0,type: int32,state_interface: velocity}  # Velocity actual value, 已修改
  #     - {index: 0x6077,sub_index: 0,type: int16,state_interface: effort}  # Torque actual value,在现有的配置文件中没有找到
      
  - index: 0x1a01
    channels:
      - {index: 0x6041, sub_index: 0, type: uint16}  # Status word，已修改
      - {index: 0x6061, sub_index: 0, type: int8}  # Mode of operation display，已修改
      - {index: 0x6064,sub_index: 0,type: int32,state_interface: position}  # Position actual value, 已修改
      - {index: 0x606c,sub_index: 0,type: int32,state_interface: velocity}  # Velocity actual value, 已修改
      # - {index: 0x6077,sub_index: 0,type: int16,state_interface: effort}  # Torque actual value,在现有的配置文件中没有找到
      # - {index: 0x60f4,sub_index: 0,type: int32}  # Following Error Actual Value
      - {index: 0x603f,sub_index: 0,type: uint16}  # Error Code
      
  # - index: 0x1a02
  #   channels:
  #     - {index: 0x6041, sub_index: 0, type: uint16}  # Status word，已修改
  #     - {index: 0x6064,sub_index: 0,type: int32,state_interface: position}  # Position actual value, 已修改
  #     - {index: 0x606c,sub_index: 0,type: int32,state_interface: velocity}  # Velocity actual value, 已修改
  #     - {index: 0x6077,sub_index: 0,type: int16,state_interface: effort}  # Torque actual value,在现有的配置文件中没有找到
  #     - {index: 0x60b9,sub_index: 0,type: uint16}  # Touch Probe Status
  #     - {index: 0x60ba,sub_index: 0,type: int32}  # Touch Probe 1 Position Positive Value
  #     - {index: 0x60fd,sub_index: 0,type: uint32} # Digital inputs

[... 1 lines cut ...]
  #   channels:
  #     - {index: 0x6041, sub_index: 0, type: uint16}  # Status word，已修改
  #     - {index: 0x6064,sub_index: 0,type: int32,state_interface: position}  # Position actual value, 已修改
  #     - {index: 0x606c,sub_index: 0,type: int32,state_interface: velocity}  # Velocity actual value, 已修改
  #     - {index: 0x6077,sub_index: 0,type: int16,state_interface: effort}  # Torque actual value,在现有的配置文件中没有找到
  #     - {index: 0x60b9,sub_index: 0,type: uint16}  # Touch Probe Status
  #     - {index: 0x60ba,sub_index: 0,type: int32}  # Touch Probe 1 Position Positive Value
  #     - {index: 0x60f4,sub_index: 0,type: int32} # Following Error Actual Value

      # - {index: 0x6061, sub_index: 0, type: int8}  # Mode of operation display，已修改
      # - {index: 0x60f4, sub_index: 0, type: int8}  # following error actual value，代码里没有，文档里有
      # - {index: 0x603f, sub_index: 0, type: int8}  # error code，代码里没有，文档里有
[... cut ...]
  - {index: 0, type: output, pdo: ~, watchdog: disable}  # MBoxOut
  - {index: 1, type: input, pdo: ~, watchdog: disable}   # MBoxIn
  - {index: 2, type: output, pdo: rpdo, watchdog: disable}  # Outputs
  - {index: 3, type: input, pdo: tpdo, watchdog: disable}  # Inputs
```

This is my config file,and I use the motor in CSP mode,I have added the 0x6061,but I still met the problem that the motor rapidly goes to position 0.And the delay time I have tried caused a bigger problem,the motor was twitching!!