ICube-Robotics/ethercat_driver_ros2 #160: slave on op mode can not control moter

11wanqin, 2025-03-07:
Ethercat enters OP mode from the slave, but the motor cannot move。

[image]

[image]

[image]

467480903, 2025-03-11:
check the motor's current position by igh command:

ethercat upload --position 0 --type int32 0x6064 0x00 #查看位置

check if the motor is in position 10000 already , 

or you can check the position by   rostopic echo /joints_status , I fogert the topic name , some like this.

and send the command once , don't send it cyclely,

11wanqin, 2025-03-11:
[image]
[image]
[image]
[image]
 the motor  position  is not on 10000 . #没有在1000
/joints_status positions is change #位置值一直在变化

11wanqin, 2025-03-11:
vendor_id: 0x00000009
product_id: 0x00009253
assign_activate: 0x0300 # DC Synch register
auto_fault_reset: false # true = automatic fault reset, false = fault reset on rising edge command interface "reset_fault"
sdo: 
 - { index: 0x6060, sub_index: 0, type: int8, value: 0 } 
 - { index: 0x6060, sub_index: 0, type: int8, value: 8 } 
 - { index: 0x6040, sub_index: 0, type: int16, value: 6 } 
 - { index: 0x6040, sub_index: 0, type: int16, value: 7 } 
 - { index: 0x6040, sub_index: 0, type: int16, value: 15 } 
# sdo: # sdo data to be transferred at drive startup
#   - { index: 0x1C32, sub_index: 0, type: int16, value: 2 } 
#   - { index: 0x1C33, sub_index: 0, type: int16, value: 2 } 
rpdo: # RxPDO = receive PDO 4 Mapping
  - index: 0x1601
    channels:
      - { index: 0x60FF, sub_index: 0, type: int32,command_interface: velocity, default: 500 } # Control word
      - { index: 0x607A,  sub_index: 0,type: int32,command_interface: position, default: .nan} # Target position
      - { index: 0x6071, sub_index: 0, type: int16, default: 0 } # Target velocity
      - { index: 0x6040, sub_index: 0, type: uint16,command_interface: controlword, default: 0} # Target torque
      - { index: 0x6081, sub_index: 0, type: int16, default: 8 } # Mode of operation
      - { index: 0x6060,    sub_index: 0, type: int8, default: 0 } # Mode of operation
      - { index: 0x0000,    sub_index: 0, type: int8, default: 0 } # Mode of operation
tpdo: # TxPDO = transmit PDO 4 Mapping
  - index: 0x1a00
    channels:
      - { index: 0x6064, sub_index: 0, type: int32 , state_interface: position } # Status word
      - { index: 0x6077, sub_index: 0, type: int16} # Position actual value
      - { index: 0x603F, sub_index: 0, type: uint16 } # Torque actual value
      - { index: 0x6041, sub_index: 0, type: uint16 } # Mode of operation display
      - { index: 0x6020, sub_index: 3, type: int16 , state_interface: effort} # Digital Input Functionalities State
      - { index: 0x6061,    sub_index: 0, type: int8, default: 0 } # Mode of operation
      - { index: 0x0000,    sub_index: 0, type: int8, default: 0 } # Mode of operation
  - index: 0x1a02
    channels:
      - { index: 0x6020, sub_index: 1, type: int16 } # Status word
      - { index: 0x6020, sub_index: 2, type: int32 } # Position actual value
      - { index: 0x6000, sub_index: 2, type: int32 } # Torque actual value
      - { index: 0x6000, sub_index: 3, type: int32 } # Mode of operation display
#this is my config file 我的配置文件

467480903, 2025-03-11:
must wait until the servo's mode in running mode  , 
then send the ros topic command  to make the servo run to some position, 

during the ininiting process, there are some consoles woudl displayed , 

you can paste these ros output here , to  disgnositc well

11wanqin, 2025-03-11:
qin@qin:~$ ros2 launch ethercat_motor_drive motor_drive.launch.py --debug
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/qin/.ros/log/2025-03-11-18-01-08-508771-qin-8797
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f1cff1aae30>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f1cff1aae30>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f1cfda93c40>'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> result=None created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 0.160 seconds
[INFO] [ros2_control_node-1]: process started with pid [8800]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [robot_state_publisher-2]: process started with pid [8802]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [spawner-3]: process started with pid [8804]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [spawner-4]: process started with pid [8806]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7f1cfd803700>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7f1cfd7e8c10>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7f1cfd7ea8c0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7f1cfd7ea380>'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f1cfd803700>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f1cfd803700>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7f1cfd802530>'
[robot_state_publisher-2] [INFO] [1741687268.696512308] [robot_state_publisher]: got segment world
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f1cfd803700>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f1cfd803700>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7f1cfd800250>'
[ros2_control_node-1] [WARN] [1741687268.696961339] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1741687268.697067405] [resource_manager]: Loading hardware 'motor_drive' 
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f1cfd7ea470>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f1cfd7ea470>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7f1cfd800250>'
[ros2_control_node-1] [INFO] [1741687268.698058416] [resource_manager]: Initialize hardware 'motor_drive' 
[ros2_control_node-1] [INFO] [1741687268.698090358] [EthercatDriver]: joints
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f1cfd803700>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f1cfd803700>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7f1cfd800250>'
[ros2_control_node-1] [INFO] [1741687268.699402773] [EthercatDriver]: Got 1 modules
[ros2_control_node-1] [INFO] [1741687268.699427206] [resource_manager]: Successful initialization of hardware 'motor_drive'
[ros2_control_node-1] [INFO] [1741687268.699462790] [resource_manager]: 'configure' hardware 'motor_drive' 
[ros2_control_node-1] [INFO] [1741687268.699468761] [resource_manager]: Successful 'configure' of hardware 'motor_drive'
[ros2_control_node-1] [INFO] [1741687268.699471977] [resource_manager]: 'activate' hardware 'motor_drive' 
[ros2_control_node-1] [INFO] [1741687268.699474654] [EthercatDriver]: Starting ...please wait...
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0x7f1cfd803700>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0x7f1cfd803700>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7f1cfd800250>'
[ros2_control_node-1] {0, 0, 0x9, 0x9253, 0x60ff, 0x0}
[... cut ...]
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7f1cfd8037c0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7f1cfd8037c0>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x7f1cfd802da0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7f1cfd8037c0>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x7f1cfd802cb0>'
# this is the consoles  displayed