ICube-Robotics/ethercat_driver_ros2 #163: Beckhoff EL7221 with DS402 state machine does not start

AltinayGrass, 2025-03-21:
I am using IGH master 1.6.3 on 22.04 with humble. I am able to use same system with Epos4 drivers. I added another system to end of ethercat line and change some settings. Here is my config files.

[EL7221 slave config](https://github.com/AltinayGrass/ethercat_driver_ros2_examples/blob/main/ethercat_diff_drive/config/beckhoff_el7221-9015_slave_config.yaml)
[EL1809 input config](https://github.com/AltinayGrass/ethercat_driver_ros2_examples/blob/main/ethercat_diff_drive/config/beckhoff_el1809.yaml)
[EL2809 output config](https://github.com/AltinayGrass/ethercat_driver_ros2_examples/blob/main/ethercat_diff_drive/config/beckhoff_el2809.yaml)
[controllers.yaml](https://github.com/AltinayGrass/ethercat_driver_ros2_examples/blob/main/ethercat_diff_drive/config/controllers.yaml)
[Launch file](https://github.com/AltinayGrass/ethercat_driver_ros2_examples/blob/main/ethercat_diff_drive/launch/motor_el7221_air.launch.py)

Here is some critical screen shots and logs.

[image]
everything is fine with EPOS4

[image]
Ethercat is like OK but state machine is stack.

[image]

[debug_l1_dmesg.txt](https://github.com/user-attachments/files/19388162/debug_l1_dmesg.txt)

Should I look for hardware problem or is it just simple fault reset request?

Any suggestions?

pzanne, 2025-03-27:
Hello,

The driver status is strange, the value 104 does not seem to correspond to a state in the cia402 state machine. In your kernel logs there is a packet loss problem, as if there were a configuration problem. Is the driver set correctly? Have you tested its behaviour from twincat? 
Best
Philippe

AltinayGrass, 2025-03-28:
It is Ok with Twincat. Because it has reset button :)
I made below change to make it run;

```
  } else if ((status_word & 0b01001111) == 0b00001000) {
    return STATE_FAULT;
  }
```
```
  } else if ((status_word & 0b00001000)) {
    return STATE_FAULT;
  }
```
But I will be very happy if you share some example of auto_reset: false case. Hardware interfaces is ready but I could find the way use it;

```
root@rnd-UPX-ASL01:~/nav2_ws# ros2 control list_hardware_interfaces 
command interfaces
	gpio0/do.1 [available] [claimed]
	gpio0/do.10 [available] [claimed]
	gpio0/do.11 [available] [claimed]
	gpio0/do.12 [available] [claimed]
	gpio0/do.2 [available] [claimed]
	gpio0/do.3 [available] [claimed]
	gpio0/do.4 [available] [claimed]
	gpio0/do.9 [available] [claimed]
	joint_1/reset_fault [available] [unclaimed]
	joint_1/velocity [available] [claimed]
	joint_2/reset_fault [available] [unclaimed]
	joint_2/velocity [available] [claimed]
```