ICube-Robotics/ethercat_driver_ros2 #164: ethercat manager service call permission problem

AltinayGrass, 2025-03-25:
Hello,

When I call service with below command, I am getting permission error.
```
root@rnd-UPX-ASL01:~/nav2_ws# ros2 service call /ethercat_manager/get_sdo ethercat_msgs/srv/GetSdo {"master_id: 0, slave_position: 1, sdo_index: 0x6040, sdo_data_type: 'int16'"}
requester: making request: ethercat_msgs.srv.GetSdo_Request(master_id=0, slave_position=1, sdo_index=24640, sdo_subindex=0, sdo_data_type='int16')

response:
ethercat_msgs.srv.GetSdo_Response(success=False, sdo_return_message='Failed to upload SDO: Operation not permitted', sdo_return_value_string='', sdo_return_value=nan)
```

Any suggestion?

pzanne, 2025-03-27:
Hello,

have you tried direct access from the command :
`ethercat upload -m 0 -p 1 -t uint16 0X6040 0X0`
does it work or do you also have access problems?

AltinayGrass, 2025-03-28:
```
root@rnd-UPX-ASL01:~/nav2_ws# ethercat upload -p 1 -t int16 0x6041 0x0
0x8060 -32672
root@rnd-UPX-ASL01:~/nav2_ws# ros2 service call /ethercat_manager/get_sdo ethercat_msgs/srv/GetSdo {"master_id: 0, slave_position: 1, sdo_index: 0x6041, sdo_data_type: 'int16'"}
requester: making request: ethercat_msgs.srv.GetSdo_Request(master_id=0, slave_position=1, sdo_index=24641, sdo_subindex=0, sdo_data_type='int16')

response:
ethercat_msgs.srv.GetSdo_Response(success=False, sdo_return_message='Failed to upload SDO: Operation not permitted', sdo_return_value_string='', sdo_return_value=nan)
```
The workspace is on docker. Rootless mode is active and not using sudo command for permission required  cases.
But I just add groups option for ethercat rules, like;

`KERNEL=="EtherCAT[0-9]*", MODE="0666" , GROUP = "docker"`

And deleted build and install directory and then rebuild the ws. Still it is same.

Any suggestion?

pzanne, 2025-03-28:
Hello,

can you try to change in the file ethercat_manager/src/ethercat_sdo_srv_server.cpp the line 57:
```
-master.open(EcMasterAsync::Read);
+master.open(EcMasterAsync::ReadWrite);
```
Best
Philippe

AltinayGrass, 2025-03-28:
same behavior

pzanne, 2025-03-28:
Stupid question, but just to be sure, did you recompile the ethercat ros2 driver after making the change to the ethercat_sdo_srv_server.cpp file?

AltinayGrass, 2025-03-28:
Yes. But to be sure, I restarted the docker container, deleted the build and install folders and recompiled it. This time it worked. Thanks.

closed 2025-03-28 by AltinayGrass