ICube-Robotics/ethercat_driver_ros2 #165: Domain working counter changes in every second

AltinayGrass, 2025-04-08:
Hello,

I am trying to use icube ethercat_ros2_driver and ros controller at 500hz. I made some simple change to display cycles. And the result is good for me. 

ecmaster::readdata function is like that;

```
  // check for master and slave state change
  if (update_counter_ % check_state_frequency_ == 0) {
    checkMasterState();
    checkSlaveStates();
    clock_gettime(CLOCK_TO_USE, &startTime);
    period_ns = DIFF_NS(lastStartTime, startTime);
    lastStartTime = startTime; 
    printf("Avarage cycle:  %10u\n", period_ns/check_state_frequency_);
  }
```
I did not see average cycle on screen in every second but after some times information getting together like below;
```
[ros2_control_node-1] Average cycle:     2613006
[ros2_control_node-1] Average cycle:     2000021
[ros2_control_node-1] Average cycle:     2000051
[ros2_control_node-1] Average cycle:     1999965
[ros2_control_node-1] Average cycle:     1999950
[ros2_control_node-1] STATE: Fault with status word :8
[ros2_control_node-1] Slave: State 0x08.
[ros2_control_node-1] Slave: operational.
[ros2_control_node-1] Average cycle:     1999999
[ros2_control_node-1] STATE: Switch on Disabled with status word :96
[ros2_control_node-1] STATE: Ready to Switch On with status word :33
[ros2_control_node-1] STATE: Switch On with status word :35
[ros2_control_node-1] STATE: Operation Enabled with status word :4135
[ros2_control_node-1] Average cycle:     2000012
[ros2_control_node-1] Average cycle:     1999992
[ros2_control_node-1] STATE: Fault with status word :8
[ros2_control_node-1] Slave: State 0x04.
[ros2_control_node-1] Average cycle:     2000032
[ros2_control_node-1] Master AL states: 0x08.
[ros2_control_node-1] Slave: State 0x08.
[ros2_control_node-1] Slave: operational.
[ros2_control_node-1] Slave: State 0x08.
[ros2_control_node-1] Slave: operational.
[ros2_control_node-1] Slave: State 0x08.
[ros2_control_node-1] Slave: operational.
[ros2_control_node-1] Average cycle:     1999976
[ros2_control_node-1] STATE: Switch on Disabled with status word :96
[ros2_control_node-1] STATE: Ready to Switch On with status word :33
[ros2_control_node-1] STATE: Switch On with status word :35
[ros2_control_node-1] STATE: Operation Enabled with status word :4135
[ros2_control_node-1] Average cycle:     2000024
[ros2_control_node-1] Average cycle:     1999973
[ros2_control_node-1] STATE: Fault with status word :104
[ros2_control_node-1] STATE: Fault with status word :104
[ros2_control_node-1] STATE: Switch on Disabled with status word :96
[ros2_control_node-1] STATE: Switch on Disabled with status word :96
[ros2_control_node-1] STATE: Ready to Switch On with status word :161
[ros2_control_node-1] STATE: Ready to Switch On with status word :161
[ros2_control_node-1] Average cycle:     2000023
[ros2_control_node-1] Average cycle:     1999963
[ros2_control_node-1] Average cycle:     2000026
[ros2_control_node-1] STATE: Switch On with status word :35
[ros2_control_node-1] STATE: Switch On with status word :35
[ros2_control_node-1] STATE: Operation Enabled with status word :4135
[ros2_control_node-1] STATE: Operation Enabled with status word :4135

```
some trigger can cause printout. This is another issue but not much important.

The important point is, I have disabled checkdomainstate function but on dmesg out there is info about domain working counter changes. There is also no missing datagrams.

```
[13120.589399] EtherCAT 0: Domain 0: 335 working counter changes - now 9/9.
[13121.593332] EtherCAT 0: Domain 0: 334 working counter changes - now 9/9.
[13122.597343] EtherCAT 0: Domain 0: 335 working counter changes - now 7/9.
[13123.601235] EtherCAT 0: Domain 0: 335 working counter changes - now 9/9.
[13124.605211] EtherCAT 0: Domain 0: 334 working counter changes - now 9/9.
[13125.609164] EtherCAT 0: Domain 0: 335 working counter changes - now 7/9.
[13126.613098] EtherCAT 0: Domain 0: 335 working counter changes - now 9/9.
[13127.617059] EtherCAT 0: Domain 0: 334 working counter changes - now 9/9.
[13128.621014] EtherCAT 0: Domain 0: 335 working counter changes - now 7/9.
[13129.624983] EtherCAT 0: Domain 0: 335 working counter changes - now 9/9.
[13130.628931] EtherCAT 0: Domain 0: 334 working counter changes - now 9/9.
[13131.632908] EtherCAT 0: Domain 0: 335 working counter changes - now 7/9.
[13132.636863] EtherCAT 0: Domain 0: 335 working counter changes - now 9/9.
[13133.640790] EtherCAT 0: Domain 0: 334 working counter changes - now 9/9.
[13134.644772] EtherCAT 0: Domain 0: 335 working counter changes - now 7/9.
[13135.648707] EtherCAT 0: Domain 0: 335 working counter changes - now 9/9.
[13136.652648] EtherCAT 0: Domain 0: 334 working counter changes - now 9/9.
```

Every single second this information logged. This is my state change check period. Is it such a coincidence or it is related with igh ethercat master?

pzanne, 2025-04-10:
Hello,

Each slave reading/writing the ethercat frame increments the counter. A partial counter change like this means that there is one slave that is not writing (+2 per write to counter) when it should be or 2 slaves that are not reading (+1 per read) when they should be. 
Which slaves are you using?  It may be related to a synchronization problem. Could you send me your configuration files so that I can see if I see anything?
According to your logs, you're using 3 drivers. Could you see if the problem also occurs with 1 and 2 drivers?
Best
Philippe

kouge0510, 2025-04-18:
> Hello, The slaves;
> 
> ```
> 0  0:0  PREOP  +  EK1100 EtherCAT-Koppler (2A E-Bus)
> 1  0:1  PREOP  +  EL7221-9015 1Ch. DS402 Servo motor output stage with OCT (50V, 
> 2  0:2  PREOP  +  EL7221-9015 1Ch. DS402 Servo motor output stage with OCT (50V, 
> 3  0:3  PREOP  +  EL1809 16K. Dig. Eingang 24V, 3ms
> 4  0:4  PREOP  +  EL2809 16K. Dig. Ausgang 24V, 0.5A
> ```
> 
> I am not sure about configuration file but I attached slave config files.
> 
> [beckhoff_el7221-9015_slave_config.txt](https://github.com/user-attachments/files/19686905/beckhoff_el7221-9015_slave_config.txt) [beckhoff_el2809.txt](https://github.com/user-attachments/files/19686906/beckhoff_el2809.txt) [beckhoff_el1809.txt](https://github.com/user-attachments/files/19686907/beckhoff_el1809.txt)
> 
> Are these correct ones?

If it works properly,you should find the status turns to be OP,maybe you should check the state machine of your motors at the time you met the synch problem.

kouge0510, 2025-04-18:
> Hello, The slaves;
>
> ```
> 0  0:0  PREOP  +  EK1100 EtherCAT-Koppler (2A E-Bus)
> 1  0:1  PREOP  +  EL7221-9015 1Ch. DS402 Servo motor output stage with OCT (50V,
> 2  0:2  PREOP  +  EL7221-9015 1Ch. [...]

Also,I met a similar problem maybe caused by the same thing,my motor only supports 125hz,but I use the program without ros2,it can up to 500hz