ICube-Robotics/ethercat_driver_ros2 #166: Can't connect to my motor

kouge0510, 2025-04-11:
hi,I met a problem when controlling my own motor.It shows

```
[spawner-5] [INFO] [1744335508.856551598] [spawner_position_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-3] [INFO] [1744335508.856553714] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-4] [INFO] [1744335508.856560570] [spawner_position_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
```

## this is my launch file

```
import os
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import ExecuteProcess
from launch.substitutions import Command, LaunchConfiguration
from launch_ros.substitutions import FindPackageShare

def generate_launch_description():
    # Get the URDF file path
    pkg_share = FindPackageShare(package='igh_motor').find('igh_motor')
    urdf_file = os.path.join(pkg_share, 'urdf', 'left_arm_3.urdf')

    # Set a parameter 'use_sim_time' to true
    use_sim_time = LaunchConfiguration('use_sim_time', default='true')

    # Read the URDF file and store it in robot_description
    with open(urdf_file, 'r') as file:
        robot_description = file.read()
    controller_yaml_path = os.path.join(pkg_share, 'config', 'controller_config.yaml')
    # Encapsulate as key-value pair for passing as a parameter to the node
    robot_description_param = {'robot_description': robot_description}

    return LaunchDescription([
        # Load the URDF file and publish it to the /robot_description topic
        Node(
            package="robot_state_publisher",
            executable='robot_state_publisher',
            output='screen',
            parameters=[robot_description_param],
        ),
        # Start the ros2_control_node and load the hardware and controller configuration files
        Node(
            package="controller_manager",
            executable='ros2_control_node',
            parameters=[controller_yaml_path],
            output='screen',
        ),
        # Start the joint_state_broadcaster controller
        Node(
            package="controller_manager",
            executable="spawner",
            arguments=["joint_state_broadcaster", "--controller-manager", "/controller_manager"],
            output='screen',
        ),
        # Start the position_joint_trajectory_controller controller
        Node(
            package="controller_manager",
            executable="spawner",
            arguments=["position_joint_trajectory_controller"],
            output='screen',
        ),
        # 启动 position_controller
        Node(
            package="controller_manager",
            executable="spawner",
            arguments=["position_controller"],
            output="screen"
        ),
        
    ])
```

## and this is my controller file

```
controller_manager:
  ros__parameters:
    update_rate: 100
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster  # 播报器类型，由ros2 controller提供的标准类型
      
    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    position_controller:
      type: position_controllers/JointGroupPositionController

joint_trajectory_controller:
  ros__parameters:
      joints:
        - L1
        - L2
        - L3
        - L4
        - L5
        - L6
        - L7

      command_interfaces:
        - position
        - reset_fault
      state_interfaces:
        - position
        - velocity
        - effort
      allow_partial_joints_goal: true #允许部分关节目标
      open_loop_control: true

     
position_controller: # 位置控制器
  ros__parameters:
    joints:
      - L1
      - L2
      - L3
      - L4
      - L5
      - L6
      - L7
      

joint_state_broadcaster: # 关节状态播报器
  ros__parameters:
    extra_joints: []
    interfaces: []  # 留空则表示所有接口
    joints:
      - L1
      - L2
      - L3
      - L4
      - L5
      - L6
      - L7
    map_interface_to_joint_state:
      effort: effort
      position: position
      velocity: velocity
    use_local_topics: false
```

I also add the     ```
<joint name="L7" type="revolute">
      <state_interface name="position"/>
      <state_interface name="velocity"/>
      <state_interface name="effort"/>
      <command_interface name="position"/>
      <command_interface name="reset_fault"/>
      <ec_module name="KDE controller">
        <plugin>ethercat_generic_plugins/EcCiA402Drive</plugin>
        <param name="alias">6</param>
        <param name="position">0</param>
        <param name="mode_of_operation">8</param>
        <param name="slave_config">../config/slave_config.yaml</param>
      </ec_module>
    </joint>
```
like this,but it still can't connect to my motor.
Do I need to start the ethercat before I use the package?
If the contents below are right,is there any possible that I wrote the wrong motor arguments?
If someone could help me,appreciate a lot!

AltinayGrass, 2025-04-11:
Yes you need to start the ethercat before. something like 

`sudo /etc/init.d/ethercat start`

Also need info about slave_config.yaml? Can you share it?

kouge0510, 2025-04-11:
> Yes you need to start the ethercat before. something like
> 
> `sudo /etc/init.d/ethercat start`
> 
> Also need info about slave_config.yaml? Can you share it?

thanks a lot,I have solved it!

closed 2025-04-11 by kouge0510