ICube-Robotics/ethercat_driver_ros2 #167: interface problem

kouge0510, 2025-04-11:
hi, i met a problem with the interface.
this is my controller.yaml
```
controller_manager:
  ros__parameters:
    update_rate: 100
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster  # 播报器类型，由ros2 controller提供的标准类型
      
    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

joint_trajectory_controller:
  ros__parameters:
      joints:
        - L1
        - L2
        - L3
        - L4
        - L5
        - L6
        - L7

      command_interfaces:
        - position

      state_interfaces:
        - position
        - velocity
      state_publish_rate: 50
      command_publish_rate: 500 # 控制器发布命令的频率

```
It told me that I couldn't use the interface shown in the documents for this package
`      
<state_interface name="position"/>
      <state_interface name="velocity"/>
      <state_interface name="effort"/>
      <command_interface name="position"/>
      <command_interface name="reset_fault"/>
`
like 

```
  command_interfaces:
        - position
        - reset_fault
      state_interfaces:
        - position
        - velocity
        -   effort
```
When I tried the  code above,it errors,so what is the correct version of controller.yaml if I want to use these interfaces?
If anyone can help,appreciate a lot!

AltinayGrass, 2025-04-11:
are you sure about configuration file of slaves?

kouge0510, 2025-04-11:
> are you sure about configuration file of slaves?

the present status is running properly

AltinayGrass, 2025-04-13:
> > are you sure about configuration file of slaves?
> 
> the present status is running properly

Congratulations

closed 2025-04-16 by kouge0510