ICube-Robotics/ethercat_driver_ros2 #169: Long Startup Time with Multiple CiA402 Drives

JinbaoXiong, 2025-04-28:
Hello,

I'm experiencing very long startup times when launching a system with multiple CiA402 drives. With 19 drives in my setup, the system takes approximately 1.5 minutes to complete initialization and reach the "Operation Enabled" state for all drives. This initialization time increases with the number of drives.

## Issue Details
- From logs, I can see it takes significant time for drives to transition through all states (Switch on Disabled → Ready to Switch On → Switch On → Operation Enabled)
- Each drive seems to be initialized one after another rather than simultaneously
- Many PDO entries are printed during startup which adds overhead

## System Information
- Using ethercat_driver_ros2
- 19 drives with vendor_id: 0x9a, product_code: 0x30924
- ROS 2（ROS Humble）running on Ubuntu 22.04

## Question
Are there configuration options or methods to reduce this startup time? For instance:
- Parameters to optimize state transitions
- Configuration options for faster startup
- Any recommendations for systems with many drives

The long startup time is becoming a practical issue for our deployment. Any suggestions to reduce it would be greatly appreciated.

Thank you!

[log_LongWaitingTime_for_Launch.txt](https://github.com/user-attachments/files/19933638/log_LongWaitingTime_for_Launch.txt)

kouge0510, 2025-04-29:
> Hello,
>
> I'm experiencing very long startup times when launching a system with multiple CiA402 drives. With 19 drives in my setup, the system takes approximately 1.5 minutes to complete [...]

Hi,I control 7 motors,it cost about 15s to start,the motors start one by one,so I don't know whether it is normal for all the drivers to start.

gkouretas-maven, 2025-12-10:
For anyone that sees this issue - the reason that there's a "long startup time" is that the IgH master attempts to synchronize the slave during setup within a hard-coded threshold, with a maximum timeout of 5 seconds. If the master software isn't being run on a hard real-time / super performant system it will take a lot longer than 5 seconds for things to synchronize (check the `syslog` dumps of the IgH master to see this). 

If DC sync is disabled this bottleneck goes away, but with it enabled improvements to the system are necessary (either by using a more performant master software or even re-configuring the IgH master stack via some compiler options to have more accurate timing).