ICube-Robotics/ethercat_driver_ros2 #171: Getting stuck in PREOP mode, because PREOP + ERROR

orjules, 2025-06-04:
Hey,

i am trying to use a [dunker BGxx dPro EC motor](https://www.dunkermotoren.de/-/media/project/oneweb/oneweb/dunkermotoren/downloads/pdf/products/001_-brushless-dc-motors/026_bg_95_dpro_pn-ec-ei.pdf?la=pt-br&revision=b258f493-f10d-4bff-8234-5eb0f4f69333) in its velocity mode (without ramps). That means i am using the CiA 402 mode 2.

To test this I created a separate project with just one joint.
The xacro looks like this:
```xml
<ros2_control name="simpleJoint" type="system">            
    <hardware>
        <plugin>ethercat_driver/EthercatDriver</plugin>
        <param name="master_id">0</param>
        <param name="control_frequency">1000</param>
    </hardware>
    
    <joint name="simpleJoint">
    <state_interface name="position"/>
    <state_interface name="velocity"/>
    <command_interface name="velocity"/>
    <ec_module name="DUNKER">
        <plugin>ethercat_generic_plugins/EcCiA402Drive</plugin>
        <param name="alias">0</param>
        <param name="position">0</param>
        <param name="mode_of_operation">2</param>
        <param name="slave_config">$(find single_joint)/config/cia_dunker.yaml</param>
    </ec_module>
    </joint>
</ros2_control>
```
The corresponding config is here: [cia_dunker.yaml.txt](https://github.com/user-attachments/files/20591208/cia_dunker.yaml.txt)
It is a slightly changed version of the [example in the docs](https://icube-robotics.github.io/ethercat_driver_ros2/user_guide/config_cia402_drive.html#usage).

When I launch everything the output in the console looks like this:
```
[ros2_control_node-2] [INFO] [1749038308.820002875] [EthercatDriver]: Starting ...please wait...
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x6040, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x607a, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x60ff, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x6071, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x60b0, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x60b1, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x60b2, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x6060, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x6041, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x6064, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x606c, 0x0}
[ros2_control_node-2] {0, 0, 0x257, 0x1, 0x6061, 0x0}
[robot_state_publisher-1] [INFO] [1749038308.840936724] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1749038308.841007530] [robot_state_publisher]: got segment simple_link
[ros2_control_node-2] [INFO] [1749038308.841125203] [EthercatDriver]: Activated EcMaster!
[spawner-3] [INFO] [1749038309.014798197] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-4] [INFO] [1749038309.022540261] [spawner_simple_position_controller]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-2] 1 slave(s).
[ros2_control_node-2] Master AL states: 0x02.
[ros2_control_node-2] Link is up.
[ros2_control_node-2] Slave: State 0x02.
[ros2_control_node-2] Slave: online.
[ros2_control_node-2] STATE: Not Ready to Switch On with status word :0
[ros2_control_node-2] [INFO] [1749038309.841296226] [EthercatDriver]: updated!
[ros2_control_node-2] [INFO] [1749038309.841315767] [EthercatDriver]: System Successfully started!
```

If I understand correctly, the line with `STATE: Not Ready to Switch On with status word :0` is the most important, because it tells me that I am somehow stuck in that state.
When I read out the status register with `ethercat upload 0x6041 0x00 --type uint16` I get  `0x0260`, meaning the "sod: switch on disabled" is 1.

This also shows itself in the fact that I cannot tell the motor to go into SAFEOP or OP from the command line, after I launched it once.
Only after resetting the motor, it is possible again.

To get more info I looked into the dmesg output.
Interestingly there is a line about not being able to set the pdos:
```
[17613.916173] EtherCAT ERROR 0-0: SDO abort message 0x08000021: "Data cannot be transferred or stored to the application because of local control".
[17613.916175] EtherCAT WARNING 0-0: Failed to clear PDO mapping.
[17613.916176] EtherCAT WARNING 0-0: Currently mapped PDO entries: (none). Entries to map: 0x6040:00/16 0x607A:00/32 0x60FF:00/32 0x6071:00/16 0x60B0:00/32 0x60B1:00/32 0x60B2:00/16 0x6060:00/8
[17613.916182] EtherCAT WARNING 0-0: Failed to configure mapping of PDO 0x1603.
```

But if i run it again, those failed mappings seem to be in there already:

```
[17614.476161] EtherCAT ERROR 0-0: SDO abort message 0x08000021: "Data cannot be transferred or stored to the application because of local control".
[17614.476162] EtherCAT WARNING 0-0: Failed to clear PDO mapping.
[17614.476163] EtherCAT WARNING 0-0: Currently mapped PDO entries: 0x6064:00/32 0x6044:00/16 0x6041:00/16 0x60FD:00/32. Entries to map: 0x6041:00/16 0x6064:00/32 0x606C:00/255 0x6061:00/8
[17614.476169] EtherCAT WARNING 0-0: Failed to configure mapping of PDO 0x1A03.
```

The full dmesg can be found here [dmesg.txt](https://github.com/user-attachments/files/20591170/dmesg.txt), note that the file includes the following actions:
1. reset
2. launch once
3. launch again
4. command state OP

The last action resulted in the message: `[17625.595899] EtherCAT ERROR 0-0: Failed to set SAFEOP state, slave refused state change (PREOP + ERROR).` 

My is probably actually super simple, because I am not familiar with Ethercat, but it cost me 3 days of trial and error already, so I though I would just ask about it.
Any help is appreciated.

jintaoChan, 2025-06-13:
It' ok that terminal shows `STATE: Not Ready to Switch On with status word :0`. Slaves may enter OP and be enabled if PDO configuration is correct. According to your dmesg, it seems your PDO configuration goes wrong. I guess your slaves does not support modifying 0x1603/0x1a03. You may check your ESI file for further information.

orjules, 2025-06-20:
Thanks for the answer. I finally got it working.
As you said, I looked into the ESI file again and changed the PDO mappings accordingly. This revealed some problems with the distributed clock, which I simply turned of by setting `assign_activate: 0x0000`. For my application this is fine.

For anyone interested, the final working yaml file for the CiA 402 mode 3 (Profile Velocity) was as follows:

```yaml
vendor_id: 0x00000257  
product_id: 0x00000001 
assign_activate: 0x0000 # not DC
auto_fault_reset: false
  
sdo:
  - {index: 0x6040, sub_index: 0, type: uint16, value: 0x80}  # Fault reset
  - {index: 0x6060, sub_index: 0, type: int8, value: 3}  # Mode of Operation: Profile Velocity Mode

rpdo:
  - index: 0x1601  
    channels:
      - {index: 0x60FF, sub_index: 0, type: int32, default: 0, command_interface: velocity}  # Target Velocity
      - {index: 0x6040, sub_index: 0, type: uint16, default: 0}  # Controlword
      - {index: 0x60FE, sub_index: 1, type: uint32, default: 0}  # Digital Outputs

tpdo:
  - index: 0x1a03  
    channels:
      - {index: 0x6064, sub_index: 0, type: int32, state_interface: position}  # Position actual value
      - {index: 0x6044, sub_index: 0, type: int16, state_interface: velocity}  # Velocity actual value
      - {index: 0x6041, sub_index: 0, type: uint16}  # Status word
      - {index: 0x60fd, sub_index: 0, type: uint32}  # Digital Inputs
```

closed 2025-06-20 by orjules